MEGA65/mega65-core #159: Disc chooser problems - nexys4ddr-widget-20200207.14-unstab-0e58286+DIRTY

adtbm, 2020-02-09:
- when entering the image chooser for drive 0, the first time, it doesn't show any images on the SD card at all. leaving and reentering the discchosser, the second time, the images are available.....

- when hovering over an image, the image content is not shown anymore.

gardners, 2020-03-26:
The problem is that the first call to dos_readdir can fail if for some reason it thinks the directory is not a directory.  My best guess is that some data structure is getting corrupted or not initialised, or that file handles are not being properly closed before hand.  This is probably related to the long-term known strange behaviour of file handles that needs to be debugged.

adtbm, 2020-04-07:
The problem was not existent in this bitstream:
unstabl 87a27d8+DIRTY,20191117.01

it is also in the latest R2 bitstream.

adtbm, 2020-04-07:
It happens with NEXYS and R2. it did not happen witzh the above mentioned bitstream.

closed 2021-02-15 by rdpeake