MEGA65/mega65-core #161: Quantify MSSIAH cartridge boot/reset timing requirements

gardners, 2020-02-13:
The MSSIAH cartridge is quite picky about power-on timing. Even the Ultimate 64 had to tweak something so that it would work.  It also doesn't currently work on the MEGA65, due to this kind of problem.

What we wish to do, is to quantify the exact timing of the MSSIAH boot process.  This means we need to get oscilloscope traces from a real C64 with a MSSIAH plugged in, showing the 5V, /RESET, /GAME, /EXROM and A0 (as a proxy for CPU activity. Bonus points for capturing all 16 address lines) signals.  

This should be done from cold power-on, and also when the /RESET line is triggered, e.g., using a reset button.

It should be ok to gather this data on a C128 instead of a C64, assuming the MSSIAH works on a C128.  But a C64 would be the golden standard for this.

(Once we have this information, we can work to match the timing on the M65's expansion port, and hopefully, get the MEGA65 running the MSSIAH cartridge. But that's for another issue.)

adtbm, 2020-02-13:
Help request has been posted in f64
https://www.forum64.de/index.php?thread/99211-mssiah-mega65-compatibility-communtity-help-needed/

lydon42, 2022-05-29:
This is a #564

LordKromdar, 2024-12-07:
Hello Mega65 friends. I recently got one and am going to quantify the boot timing this week with my MSSIAH cart. Will respond here with info.