MEGA65/mega65-core #163: MEGA65 R2 PCB I2C peripherals not responding

gardners, 2020-02-14:
As described. No responses.

gardners, 2020-02-14:
Testing by modifying hdmidebug target to waggle the lines, to make sure they are correctly hooked up.
The weird thing is, it has previously worked, as I saw the RTC refusing to set the time when written to one byte of the clock at a time.  But now, it seems to just be stuck high.

gardners, 2020-02-14:
Ok, now we can see the registers for the HMAC/random bytes stuff, as well as the RTC registers, including the RTC SRAM.  RTC SRAM is working for being written to.

gardners, 2020-02-17:
Remaining problem is that we can't write to the RTC time/date registers.  All other RTC registers can be written to. This means we are likely not sending the STOP condition at the end of the write, which is specifically required for the RTC time/date registers, to prevent invalid writes from taking effect.

gardners, 2020-02-17:
Simulation confirms that we weren't sending the STOP signal.  Fixed (hopefully) in the above. Should also be done for other I2C busses, as this is a likely cause of some of the weird things we have seen.  Copy changes once confirmed working here.

gardners, 2020-02-18:
Not properly fixed.  The issue is that we need to allow the I2C bus to go idle after doing the read-all-regs loop, before starting any queued write. But the state machine doesn't handle this, so I'll need to add some mechanism to delay activation of command requests.

gardners, 2020-02-18:
Confirmed I2C write begins and ends with STOP/START sequences in waveform of the above commit.  Now synthesising to test...

gardners, 2020-02-18:
Confirmed that this fixes writing to the RTC clock, and setting it going.

Now we should do the same fix to the other I2C busses for the various targets.

gardners, 2020-02-18:
@frehwagen this one is safe for merging onto stable_base when you are ready.  There are a few others I have closed recently that should also be good to check for merging in.

frehwagen, 2020-02-21:
I reopen this issues because we want to have issues open until they are completed, fully tested and merged into stable_base branch.

adtbm, 2020-03-15:
**Testing 163:**

- **HDMI** NTSC -OK- / PAL -OK-
- **VGA** NTSC -OK- PAL -distorted-
[image]

- **FREEZER MENU**  THUMBNAIL -OK- Frame -NOT OK- (missing)
[image]

- **SOUND** -No Sound via HDMI NTSC from 3.5mm Audio jack
                     - Sound OK via HDMI PAL from 3.5mm Audio jack

- **SPRITES** - heavy "ghosting" with multiplexed Sprites
[image]

- **GEOS** HDMI NTSC -OK- (very good)
                  HDMI PAL -OK
                  FREEZER menu -NOT VISIBLE (blue screen (thumbnail visible, nothing else), but 
                  operational (i.e. Pressing "V" to change video mode or "F3" to leave menu.
[image]

                   RTC tool (MEGA RTC) is working, crashes GEOS but after power recycle date & time are
                   still set.

bluewaysw, 2020-03-15:
As for the SPRITES issues, what this correct when using stable_base bitstream?
FREEZER currently is not able to freeze GEOS, but this should be a nother story/issue.
Not sure with the SOUND, I think this relates to FREEZER version beeing used.

gardners, 2020-03-16:
The lack of frame around thumbnail may be missing C64THUMB.M65 etc from SD card.

frehwagen, 2020-03-16:
The functionality expected here and resulting bitstream has been successfully tested, merged back to stable_base branch so this issues is considered closed now.

closed 2020-03-16 by frehwagen