MEGA65/mega65-core #164: MEGA65 R2 PCB external microSD card slot not working

gardners, 2020-02-14:
As described.

gardners, 2020-02-14:
I can waggle all lines EXCEPT D3/CD, which is required to bring the card to attention.  This is presumed to be the root cause of the problem.

gardners, 2020-02-14:
The D3/CD should be pin K2 according to the schematics. This is also connected to the M_RX pin on the MAX10.

gardners, 2020-02-14:
Probably a good idea to try waggling it from the MAX10.  Problem with that, is that for some reason I can't get the write.sh or flash.sh scripts to run. Even after de-configuring the Xilinx FPGA so that its DONE signal clears, allowing the MAX10 JTAG to enable, and making sure the MAX10 FPGA is visible to fpgajtag, the quartus tools are unable to find it.

gardners, 2020-02-15:
ok, so confirmed that the pin is K2, by using the JTAG boundary scan feature of monitor_load to watch while grounding the pin.  This just leaves the question of why the FPGA isn't able to pull the pin down.  Is it be cause the MAX10 is not tri-stating the pin?

gardners, 2020-02-15:
One way to see if this is the case, is to tell the FPGA to drive the pin with higher current, which I'll try. But I'm not yet too confident that it will solve the problem, because I would expect to at least see a ripple on the signal, if the drive was insufficient.

gardners, 2020-02-15:
ok, so it needs to be 24mA to work.  Now synthesising for M65R2.

gardners, 2020-02-15:
Confirming that external SD card now works.

closed 2020-02-15 by gardners

gardners, 2020-02-18:
@frehwagen feel free to merge in to stable_base when you are happy with the changes.

frehwagen, 2020-04-09:
Fixed as part of #156, merged into stable_base/14.