MEGA65/mega65-core #165: Get HyperRAM working on MEGA65 R2 PCB

gardners, 2020-02-16:
We have never gotten it working.
Adding logic to slow down transactions to ~Hz speed, so that we can monitor what is happening with the JTAG boundary scan stuff, since there are no pins accessible.

gardners, 2020-02-16:
Datasheet on the Hyperram is here: http://www.issi.com/WW/pdf/66-67WVH8M8ALL-BLL.pdf

Access protocol should be quite simple and straightforward.  Read operations are described in Figure 1.1 on page 4 of the datasheet.

gardners, 2020-02-16:
despite trying a pile of things, all the hyperram pins are apparently logic 0.  This is very suspicious, given some are inputs to the hyperram. I'll need to figure out what the problem is here. Is it a powered down IO bank on the FPGA?  Stupid VHDL errors on my part?

gardners, 2020-02-16:
so, hr_* lines all act as though tied to GND. Made sure it wasn't my logic driving them at fault by switching with LED pin, which then works fine.

gardners, 2020-02-16:
Bank 16 should be powered with 3.3V, and things like the joysticks etc are on the same bank, so assume the bank isn't the problem.

gardners, 2020-02-17:
ok, so I can talk to the HyperRAM on the TE0725-03 -- even though I don't seem to be able to write reliably to it.  JTAG shows activity again. This at least proves that my HyperRAM driver is capable of life.  So now to resynthesise for the M65R2, and see if we see any life there.  If not, then assume pins must be incorrect or some other fault on the PCB layout for the hyperram?
Will slow down the HyperRAM accesses again and set a pattern on hr_d (see commit 5e76e15 above) and try running on both M65R2 and TE0725-03 PCBs.

gardners, 2020-02-18:
Consulted with Antti who confirmed pinout is correct, and that he has had the hyperram working on his board, using litex: https://github.com/micro-FPGA/litex-boards/blob/master/litex_boards/partner/platforms/mega65.py

So why are thes pins staying low for me on the MEGA65R2 board, no matter what I do, but using much the same driving code, I can see the lines toggle on the TE0725-03 board?

gardners, 2020-02-18:
Confirmed M65R2 HyperRAM is 3V, and thus setup correctly. The HyperRAM on the TE0725 is actually a totally different part, with 1.8V IO.  Thus it isn't an exact proxy for the M65R2 part testing.

Last commit above is to enable monitor to try to read from HyperRAM which we currently have clocked super slow, so that we should be able to examine the signals via JTAG boundary scan.

(As well as verifying the M65R2 board behaviour, we can also get the HyperRAM more completely working on the TE0725 board, via the same JTAG testing).

gardners, 2020-02-18:
Just confirmed that a bitstream from Antti works, even though the JTAG boundary scan still shows 0 on all the HyperRAM pins. So it looks like I can't really trust the JTAG probing of the HyperRAM.

It might well be, that when I debug the HyperRAM on the TE0725, that it will then work when transferred to the MEGA65 R2 target.  I just need to think of a way to be able to watch and debug what is going on.

gardners, 2020-02-18:
Begun tackling problem by implementing a fake hyperram model that I can test against, now that I know the JTAG boundary scan stuff is not working for whatever unknown reason. This has already turned up several bugs in the hyperram controller.  Will continue with this approach.

gardners, 2020-02-18:
On that note: known things I have to fix so I don't forget:

1. Writing odd/even bytes.
2. Check reading logic for odd/even bytes also.
3. In model, check for RWDS going low ahead of writing actually beginning, then do actual writes.

gardners, 2020-02-19:
With the above commits, the test harness can now read and write.  Test via synthesis comes next.

gardners, 2020-02-25:
Added last commit, because even with the bitbashing debug registers which should always answer, the slow_devices wrapper interprets the RAM as having timed out.  To debug this, I have made it so that the bitbashing registers are ALWAYS read in the hyperram, and asserts the data_ready_strobe thus on any read of the hyperram.  If this works, then maybe the address comparison for the debug registers is wrong. If it doesn't work, then there is some nasty plumbing issue.

gardners, 2020-02-26:
Hyperram never seems to be readable, even with fake data source.

gardners, 2020-02-28:
Okay, with cache turned off, and using internal HyperRAM, we now seem to be able to *mostly* read from the HyperRAM.

There are some gotchas, though:

1. If RAM > 8MB is attempted to be accessed, then reads below 8MB can fail until an attempt is made to write. but it's not consistent. e.g.:

M8000000 <- read from 0MB point, returns lots of random values
M8800000 <= read from 8MB point, i.e., from 2nd (non-present) hyperram die. Returns regular pattern.
M8000000 <- read from 0MB point, should return lots of random values, but sometimes just repeats the pattern from M8800000

2. More importantly, it looks like we read the previously presented byte at all times, e.g.:

m8000000
:08000000:9405514A806F40D0D0191A106C035761

.m8000000
:08000000:6105514A806F40D0D0191A106C035761

.m8000001
:08000001:61514A806F40D0D0191A106C03576161

.m8000001
:08000001:61514A806F40D0D0191A106C03576161

.m8000002
:08000002:614A806F40D0D0191A106C0357616173

.m8000002
:08000002:734A806F40D0D0191A106C0357616173

This would seem to be a problem with the latching of the data read by hyperram.vhdl into slow_devices.vhdl, not a problem with actually reading the HyperRAM itself.  This might be possible to identify the cause of through simulation.

3. Writing seems to almost universally fail.  Presume that latency timing of writes is out by a cycle or two, causing no writes to get latched.

4. In fact, attempting to write can cause strange problems, e.g.:

.m8000002
:08000002:734A806F40D0D0191A106C0357616173

.s8000000 1 2 3 4 5

.m8000000
:08000000:01020304050500000004040404040404

.m8000000
:08000000:04000000000000000004040404040404

.m8000000
:08000000:04000000000000000004040404040404

Writing a single byte seems to break it out of this loop:

.s8000000 99

.m8000000
:08000000:99995549814F40D0D011191044435541

.m8000000
:08000000:41055549814F40D0D011191044435541

Presume a problem with the latching of writes when the HyperRAM interface is busy with an asynchronous write.

Note that we temporarily read back the written value twice (maybe a symptom of how the HyperRAM module gets shocked out of the confused state), but after that, the write has evidently not taken effect.

gardners, 2020-02-28:
Above commits should fix the read latency, and make it possible to adjust the write latency, to see if we can find the correct value.

gardners, 2020-04-07:
So, back at 08872b4 we have some strange behaviour happening with the ISSI hyperram:

With write_latency set to $0E:

1. Reading is fine. No problems at all that I am aware of.
2. Writing EVEN addresses, one at a time, works fine, too.
3. Writing ODD addresses results in the value being written to the correct location, as well as ADDR+2, and often ADDR-(ADDR&7)+1, especially if (ADDR&7)==7

If we set write_latency to $10, then writing to odd addresses works, but of course writes to ADDR+2 instead of ADDR.

Also, if we do multiple writes in quick succession, this seem to bork things, too. So we need a delay after doing one write, before allowing the next. This minimum turn-around time is mentioned in the specifications somewhere.

So we could do some slightly ugly thing with adding 2 to write_latency when writing odd addresses.  this seems really ugly.

So, debugging by using the external HyperRAM PMOD module thingy from 1BitSquared which has a Cypress HyperRAM on it, things are not that different:

1. When reading, we seem to need to allow an extra tick before the data actually comes, this means $xxxxxx0/8 always reads the last byte of the CA value ($00 or $04, respectively).
2. There is some funny stuff going on with duplicated by writes, which is a bit hard to pin down completely, but actually looks quite similar to what we are seeing with the ISSI part.

So it looks like writing a single oddly-addressed byte requires masking an entire word for write first, and we need to enforce enough turn-around time between writes.

gardners, 2020-04-07:
The fix for writing from one word earlier mostly works -- except when writing byte offset +$001 after KB boundary.  In that case, the byte ends up in several variable places in the 1KB area the write has occurred in, and/or in the previous 1KB.

This is presumably because we have fixed the writing of odd addresses by going back one address, and then we step over an internal row or column boundary or something in the HyperRAM's actual DRAM, which mean that things don't get time to settle, and the data can get committed to multiple memory cells.

We should be able to deal with this by using wrapped writes of a fixed size, and then just adjusting the address within that wrap range, and thus always staying on the same memory page/row/column. 

Another approach would be to collect writes on an 8 byte cache line, and then commit the whole thing with no masking, as that will also presumably work without problems.

To do this, we need to be able to read and write the config registers for the HyperRAM, which we should do anyway, as it is the best hope we have for reducing read latency in the common case when single latency will be fine, instead of being fixed to double latency.

gardners, 2020-04-07:
With commit 0c6da6c, the following seems to work:

sbfffff5 1f       <- enable proper writing of conf registers
sbfffff8 0 0     <- prepare to write $0000 into conf register
s9001000 1   <- write those values into CR0 conf register (in theory at least)
sbfffff3 6        <- reduce countdown from $e to $6

(Still with the problem with writing to offset $001 in a given KB doing weird things).

Speed test results before this change:

Cache enabled:
Extra RAM -> Fast RAM copy: 5001KB/sec
Fast RAM -> Extra RAM copy: 1970KB/sec
Extra RAM -> Extra RAM copy: 1477KB/sec
Fill Extra RAM: 1970 KB/sec
Cache off:
Extra RAM -> Fast RAM copy: 1625KB/sec
Fast RAM -> Extra RAM copy: 1970KB/sec
Extra RAM -> Extra RAM copy: 902KB/sec
Fill Extra RAM: 1970 KB/sec

Speed test results AFTER this change:

Cache enabled:
Extra RAM -> Fast RAM copy: 5417KB/sec
Fast RAM -> Extra RAM copy: 2407KB/sec
Extra RAM -> Extra RAM copy: 1757KB/sec
Fill Extra RAM: 1970 KB/sec
Cache off:
Extra RAM -> Fast RAM copy: 1912KB/sec
Fast RAM -> Extra RAM copy: 2407KB/sec
Extra RAM -> Extra RAM copy: 1083KB/sec
Fill Extra RAM: 2407 KB/sec

So we are certainly speeding up writes quite appreciably: 2407/1970 = ~22% faster aka write latency reduced to ~0.82x original.

But there is a problem writing to CR0, as it seems to get $FFFF written into it, instead of the correct value. 

Having $FFFF would give 4 cycles latency instead of 6, which then means write_latency reduces from $E to $6 (four internal ticks per cycle). It would also leave the extra latency always enabled for reading, which explains why the reading operations have not sped up at all. Also, it should be possible to drop to 3 cycle write latency to get a little extra turn of speed.

Basically it doesn't seem to matter what we try to write to CR0, it seems to get written as $FFFF.
Figure 5.1 of the ISSI HRAM datasheet indicates that the wave form should have the data immediately following.  Only difference I have spotted so far is that the linear/wrap bit should be set to linear for register writes, apparently.  If it is doing some weird wrap and overwriting with reading tri-stated pins, that could well explain what we are seeing with not setting CR0 properly.

gardners, 2020-04-07:
Minor diversion: The cache is still messing something up, as the following sequence shows:

1. Let's put some known values into $8000000:

`.s8000000 0 1 23 4 5 6 7 8 9 a b c d e f
`

2. Let's see what the current state of the cache is:

```
.mbfffff0
:0BFFFFF0:00FF030E021F07003412010000000000
```
So cache row 0 has $8000008-F, and cache row 1 has $8000000-7, with zero valid bytes in either, because the cache is disabled (bit 7 of offset $2).
(offset $0 = valids for row 1, offset $7 = valids for row 0, offsets $a-$c = address bits 26 downto 3 for row 0, offsets $d-$f = address bits 26 downto 3 for row 1).

While the cache is off, reading config registers will also stuff up, reading only the first byte, because of how config registers work on the ISSI Hyperram.  Why we see $12 and $34 as fill bytes is not totally clear to me, though.  Those are the values loaded into the buffer for writing to config registers ($BFFFFF8-9, seen above).

```
.m9000000
:09000000:0C123412341234123412341234123412
```

Again, if we check the status of things, we see the cache line offsets have no updated, but with all valid flags cleared:

```
.mbfffff0
:0BFFFFF0:00FF030E021F07003412000020010020
```

Let's also just make sure the values we wrote can be read back.  There is still the bug with writing to offset $001 in each KB, so that byte is zero, and I didn't type a space between 2 and 3, so its all shifted left a bit, but that's not a worry for us here:

```
.m8000000
:08000000:0000230405060708090A0B0C0D0E0F00
```

So let's enable the cache:

`.sbfffff2 83
`

And check that we can still read the bytes we wrote:

```
.m8000000
:08000000:0000230405060708090A0B0C0D0E0F00
```

Which we can now see are loaded into the cache, with all valid bits set, as the cache rows have been read:

```
.mbfffff0
:0BFFFFF0:FFFF830E021F07FF3412000000010000
```

Also, if we now  read the ID register 0, we get the 8 bytes of repeated values, as we expect:

```
.m9000000
:09000000:0C810C810C810C811212121212121212
```

And we see that these addresses are now happily loaded into the cache, with all valid bits set.
Well, actually, the $9000008-F gets loaded into the cache, leaving $8000008-F still in there.

```
.mbfffff0
:0BFFFFF0:FFFF830E021F07FF3412010020010000
```

This is when it starts getting strange, as if we read $9000000 again, we should just read the cached values, but instead, we get this:

```
.m9000000
:09000000:34343434343434341212121212121212
```

So, even though $9000000-7 should be getting freshly loaded, and it is indeed loaded into the cache, we get rubbish in there.  But the cache content looks file:

```
.mbfffff0
:0BFFFFF0:FFFF830E021F07FF1234010020000020
```

Now the really weird part comes when we load $8000000 again, in that the first transaction reads rubbish:

```
.m8000000
:08000000:0000000000000000090A0B0C0D0E0F00
```

I think the problem might actually be that when the HyperRAM has done a config register read (or possibly when my HyperRAM controller has done a config register read...), the next transaction just goes wonky. 

If this is the case, we should be able to reliably read $8FFFFF8-$9000007 (if cache is flushed) to get the first config register, which we seem to be able to do.  Ok, so that's that mystery solved: The cache is fine, it is just switching to and from config register operations that is crazy.

Now to finish debugging setting config register values...

gardners, 2020-04-07:
Now, back to writing to config registers:  This still seems to be rather wonky.
My best guess so far is that when I do a write, I am ticking the clock as well as releasing CS0 -- this might cause a subsequent write to get registered.  This may in fact also explain the double-writes for odd addresses that I was seeing...  So, CS should release without an extra clock tick.  But for writing to even addresses, we probably still need to give the odd tick, so that a complete word transfer has happened.

gardners, 2020-04-07:
ok, the above commits should hopefully fix a variety of evils we were experiencing, and appreciably speed up access to the hyperram in the process.  It should even fix the writing to offset $001, if I have understood things correctly.

gardners, 2020-04-08:
ok, so that has certainly helped speed things up, but now we are still getting intermittent corruption of memory when writing.  
This is usually taking the form of $BE ("before even" mask byte written to hr_d), and double-writing of the correct value at ADDR+1 and ADDR+3.  This has been seen mostly with even addresses at the moment. 

I am wondering if the latency needs to be increased from 3 cycles to 4, or whether it is glitching on the HyperRAM interface, possibly due to poor timing closure.

gardners, 2020-04-08:
Confirming setting CON0 back to $8F1F and write_latency to $0E makes it all work reliably again.

Trying now with $8F17 (just switching to variable latency), and write_latency = $09
Yes: switching to variable latency would seem to be the problem.  

Next step is to make the extra latency adjustable, so that we can get the timing exactly right when it is required.

gardners, 2020-04-08:
So while waiting for that to finish, it's time to start looking at collecting writes together to improve the write speed.

gardners, 2020-04-08:
Cached reading and writing now seems to work fine, and is quite fast:

Copies between fast and hyperram are ~8MB/sec in either direction.
Filling hyperram is also about 8MB/sec.
There is probably scope to further optimise the write scheduler, but it is a darn sight better than it was.

Remaining to be fixed, is that writing without the write scheduler enabled now doesn't work for some reason.  Should be debuggable in simulation.  But also isn't that much of a priority, since the default operation with cache on is what we want, and much more perfomant, in any case.

Apart from that, I just want to do a bit more testing.

Further improvements, e.g., prefetching data for the caches (possibly by fetching 16 bytes instead of 8, and having a "2nd row" buffer that can get moved into place when reaching the end of a cache row and/or actively prefetching a cache line and/or maintaining a larger set of "cached cache lines" that can't be searched for validity with zero latency, but with one cycle latency can be searched, or deferring the toolate signal when collecting writes to improve write performance and/or allowing chained writes of > 8 bytes can all wait.

gardners, 2020-04-08:
So, we will leave the cache on, and not worry about debugging non-cached writes.
I have started looking at shorten clock ticks when data stays the same, since in theory it should be fairly easy to do, and will reduce latency of all transactions maybe 10% or 20%

gardners, 2020-04-11:
165-hyperram feature branch has been created with the commits cherry-picked on.
Bitstream built, but the Hyperram is not writing properly, so will need to be debugged..

gardners, 2020-04-13:
So now the slow mode in this setup mostly works, except that 8th byte in each fetch sometimes gets ORd with the following byte.
E.g.: s8000040 30 31 32 33 34 35 36 37 0

will allow reliably reading the $37 from $8000047, but if we instead:

s8000040 30 31 32 33 34 35 36 37 FF

Enabling fast read doesn't change this behaviour.

Enabling fast write writes values somewhat messed up, which makes sense, as with the clock not offset by half a tick the wrong values and mask values can get latched by the HyperRAM. The same goes for the fast command mode.

So, slow reading should be  fixable by latching the data on the correct clock phase.

The fast modes will need an actually reliable way to delay the HyperRAM clock.  Maybe I should just provide a free-running clock for the HyperRAM, and just allow switching the phase of that, rather than enabling delayed or not delayed, so that there can be no glitching of the clock.

gardners, 2020-05-04:
Above changes are making good progress.
Internal hyperram seems to now be fairly stable with cache and fast read enabled.
Fast CMD and fast write still are unstable, but that's fine. They can stay disabled, as they provide relatively little benefit.

External hyperram is also getting closer to working. Oddly it gets lots of single-byte read errors if the cache is disabled. If cache and block reads are enabled, then they are rarer -- but that's probably just because it performs many less read transactions.   The main behaviour is that read transactions can get 2 bytes of rubbish at the start, presumably because a clock tick gets missed by the external hyperram.  I have enabled fast slew and 24mA drive strength to try to resolve this. Will test on synthesis.

Simulation is still revealing some corner case problems that I need to resolve, and will work on next.
In particular, linear writing to external hyperram seems to lose some writes.  This might well be related to the problem above.

gardners, 2020-05-04:
Attempting to read an address just after a pre-fetch of the same address seems to cause the state machine to get very confused, and the read request from slow_devices times out.

frehwagen, 2020-05-22:
IEC serial devices become instable in GEOS, probably because of the timing changes. This needs to be fixed.

gardners, 2020-05-22:
Yes, this is very likely the case. However, I prefer that we don't hold up this and 138 on that, and instead file a new issue for "IEC devices unstable in GEOS following CPU timing changes", and fix that specifically there.  Otherwise we have too much in one go, and it will take too long to merge it in.

lydon42, 2023-05-02:
checked with Release 0.95 core, no problems with HyperRAM found.

commit 2fef25c7 (2020-05-15): More merge fixes #165.
closed 2023-05-02 by lydon42