MEGA65/mega65-core #168: D030 / "C128 fast" mode bit should be reversed?

lgblgblgb, 2020-02-23:
I really don't know too much about C128, unfortunately. But according to Paul, the current behaviour of MEGA65 is:

"Confirming 1 = 1MHz, 0 = 2MHz"

However stated by others (as I've told, I have no idea), the **opposite** should be true! Since it seems some C64 titles uses this bit (in the hope that it's a C128 in C64 mode so they can gain some speed, and setting this bit has no effect on a real C64), this can be a problem. It was discussed on mega65 forum at forum64.

Please someone can check this? First, how it works on C128 for real. Second: is Mega65 complies to the rules, or really it's reversed behaviour now! Third: can we fix that in mega65-core (if it's really reversed!), and then in Xemu too (this one by me).

I wouldn't change it in Xemu unless it's changed in mega65-core as well for obvious reasons :)

gardners, 2020-02-23:
Yes, it has been the wrong way around since the beginning :) Fixing now.

gardners, 2020-02-23:
@frehwagen feature branch created and committed. Change also applied to unstable

adtbm, 2020-03-16:
Testing 168:

    **HDMI** NTSC -OK- / PAL -OK-

    **VGA** NTSC -OK- / PAL -distorted-
    [image]

    **FREEZER MENU** THUMBNAIL -OK- Frame -OK- 

    **SOUND** - Sound -OK- via HDMI NTSC from 3.5mm Audio jack
                 - Sound -OK- via HDMI PAL from 3.5mm Audio jack

    **GEOS** HDMI NTSC -NOT OK- (garbage picture)
              HDMI PAL -NOT OK-
              FREEZER menu -NOT VISIBLE (blue screen (thumbnail visible, nothing else), but
              operational (i.e. Pressing "V" to change video mode or "F3" to leave menu.
              RTC tool (MEGA RTC) - Can't test-
              OBSERVATION: after running GEOS, when switching MEGA65 off/on i receive this picture,
              instead of normal MEGA65 bootup. several tries of powercycling solve this.
[image]

**Comment to add:** this is the first bitstream, where multiplexed Sprites did not have any timing issue ! i am totally impressed ! WOW ! ! !

I have to cancel the above praisals.....
**C64 starts in 2MHz (Freeze menu shows 1 MHz) so YES the frequencies are inverted, but the Freeze menu doesn't displays this.
So now, when selecting 1MHz, you are running in 2MHz and vice versa.....**
At least this shows me, that there is a still a timing issue, since the demo i am using for testing was written for a 1MHz C64.
In 1MHz i still have the desynced multiplexed sprites. in 2 MHz it looks smooth..

adtbm, 2020-03-20:
issue # 168/4 Bitstreamtest
used Bitstream: mega65r2-20200317.19-HEAD_n-0ca6ef6+DIRTY.bit,
with SD-card files from subfolder.

- atm no change to previous bitstream. 1MHz <-> 2MHz Problem not solved.

deftmega, 2020-03-20:
- Switching to NTSC locks up machine in C65 mode

- Switching to NTSC in C64 mode results in top and bottom border disappearing. Hard lock-up see below.
[image]

- NTSC mode in C64 mode is super unstable. Machine will hard lock-up. Even reset button does not bring it back.
[image]

- 2 Mhz results in 1.95 in Synthmark. 1 Mhz results in 0.95 in Synthmark.
[image]
[image]

frehwagen, 2020-03-20:
Actually, from looking at the code internally (cpu-side), (1) mean 1mhz, (0) mean 2 mhz. On the way to the registers $D030 values has been inverted, so properly presented to the outside world. So in the end there is nothing to be fixed here, will close without change.

closed 2020-03-20 by frehwagen