MEGA65/mega65-core #174: Examine HDMI video sync timing differences between stable_base and unstable

gardners, 2020-02-26:
Falk has provided a bunch of screenshots showing monitor scan info for stable_base vs unstable HDMI output.  This seems to indicate that the most probable cause is difference in HSYNC and VSYNC pulse positions and widths.  Thus some investigation of these would be a good step to fixing the HDMI output problems on unstable. (Note that HDMI audio is still a whole separate kettle of fish).

gardners, 2020-02-27:
Unfortunately, examining the two bitstreams VGA output, the HSYNC and VSYNC signals seem to  identical in terms of frequency and position relative to active/inactive parts of rasters and frames.  So back to the drawing board :(

rdpeake, 2021-02-16:
closing as hdmi has been tested for compliance via testing equipment on r3 board. (r2 won't be fixed)

closed 2021-02-16 by rdpeake