MEGA65/mega65-core #175: Implement scripts for auto-generating test script document

gardners, 2020-03-01:
We want to have a semi-automatically generated test script for confirming the correct function of the MEGA65. 
We want to ensure that such documentation has full coverage of all known issues.
To do this, we will read special fields from the body of each issue on github, to generate a list of things that SHOULD get tested.  These tags will be searched for in a special latex file that will be used to generate the actual latex for the test script, which will then be turned into a PDF.  Any tags that have not been mentioned in the latex file will be listed at the end of the document as "functionality and past problems not currently covered by this test procedure".

gardners, 2020-03-01:
The format for the special fields is:

##BREAKS Description of what doesn't work.

e.g.:

##BREAKS MEGA65 fails to produce display on VGA port

These fields MUST be in the body, i.e., the first block of text in the issue (so, for example, these here won't show up).  They can be added after the event by editing the text block.

lydon42, 2024-08-17:
This is now #593

closed 2024-08-17 by lydon42