MEGA65/mega65-core #182: Loading directory and first title on 3,5" disk unstable

deftmega, 2020-03-20:
Stable_base-13:

With real 3,5" disks loading first titles from disk has become unstable, locking up the machine from time to time. Old problem loading programs ("middle" of directory and disk) persists.

rdpeake, 2021-02-27:
[drvtest.zip](https://github.com/MEGA65/mega65-core/files/6054465/drvtest.zip)

[image]

Test file and image of running the test supplied by deft.

Need to write the supplied d81 to a physical disk and then load and run the program as shown in the image.

lydon42, 2022-01-03:
Tested with pre release development core and r3 machine. Wrote files with WRITE.PRG, then tested READ both in M65 and in C64 mode. All worked will.
Will retest with r3a as soon as it arrives.

closed 2022-01-03 by lydon42