MEGA65/mega65-core #185: Flash menu: choosing slot with number keys will always load slot 1

deftmega, 2020-03-23:
When choosing a slot to boot from using number keys in flash menu, machine will always load from slot 1

gardners, 2020-04-01:
There is something weird going on here with setting the WBSTAR address, i.e., the address the FPGA will read from in the flash memory.  Basically it is not working the way it should.

Trying different WBSTAR values with a setup with bitstreams in slot 0, 3 and 5, we get the following results:

ff000000, ff800000, ffc00000, ffe00000, fff00000, fff80000, fffc0000, fffe0000, 00000000 - slot 0, without ambulance lights
ffff0000, - slot 6 after 1.5 cycles of ambulance lights
ffff8000 - slot 0 after 4.5 cycles of ambulance lights
ffffc000 - slot 0 after 1.0 cycles of ambulance lights
ffffe000 - slot 0 after 0.5 cycles of ambulance lights
c010 - slot 3, no delay
14010 - slot 5, no delay

So the problem is that the WBSTAR address is x $100, not a literal address. Easy to fix :)

gardners, 2020-04-01:
Confirmed fixed on unstable.

gardners, 2020-04-01:
Ready for merging.

gardners, 2020-04-01:
This last commit was made to address a race condition I found in reconfig.vhdl and megaflash.c, where megaflash was not waiting any time after setting the WBSTAR address in $D6C8-B, before writing to $D6CF to activate reconfiguration.  But the state mackine in reconfig.vhdl can take ~ 200 cycles to update the actual WBSTAR parameter.

frehwagen, 2020-04-09:
Fixed as part of #156, merged into stable_base/14.

closed 2020-04-09 by frehwagen