MEGA65/mega65-core #188: Interrupt immediately following JMP ($xxxx) seems to save address of JMP, not its target

gardners, 2020-03-24:
See example trace below:

.r

PC   A  X  Y  Z  B  SP   MAPH MAPL LAST-OP In     P  P-FLAGS   RGP uS IO ws h RECA8LHC
FCFF 97 01 84 00 00 01FF 0000 0000         00     A1 N.E....C ...P 10 -  00 - .....lhc
,0777FCFF  6C 00 A0  JMP   ($A000)

.

PC   A  X  Y  Z  B  SP   MAPH MAPL LAST-OP In     P  P-FLAGS   RGP uS IO ws h RECA8LHC
FF48 97 01 84 00 00 01FC 0000 0000         00     A5 N.E..I.C ...P 10 -  00 - .....lhc
,0777FF48  48        PHA   

.r

```
PC   A  X  Y  Z  B  SP   MAPH MAPL LAST-OP In     P  P-FLAGS   RGP uS IO ws h RECA8LHC
FF48 97 01 84 00 00 01FC 0000 0000         00     A5 N.E..I.C ...P 10 -  00 - .....lhc
,0777FF48  48        PHA   

.m1fc
:000001FC:97A1FFFC000000000000000000000000

```

Note that the address of the JMP at $FFFC has been pushed to the stack, instead of the address of the target of the JMP.

This is causing boot lockup (often with rubbish on the screen) sometimes.
Possible cause of the white screen problem with utility menu etc as well?

Problem manifests with a clean build of the unstable branch commit 6715006b607c9567a15a622c859109cfe9364bc0, but not with my existing build directory, for whatever reason.

gardners, 2020-03-25:
Fixed. Ready for merging in. Problem was different to suspected in the end (although it is possible that this odd behaviour with JMP indirect persists, but in a harmless manner).

frehwagen, 2020-04-09:
Merged to stable_base/14, as part of #156 feature.

closed 2020-04-09 by frehwagen