MEGA65/mega65-core #189: $D610 ASCII keyboard acceleration ignores CAPS LOCK key

gardners, 2020-03-25:
Reading keys from $D610 doesn't change the alphabetic keys when CAPS LOCK is active. It should.

##BREAKS CAPS LOCK enables capital letters in $D610 ASCII key feed, e.g., typing in FDISK

gardners, 2020-04-03:
Hmmm, this seems to have either regressed, or never actually worked.
Possibly related: pressing two keys in quick succession results in only the second one being registered.
Regular shift works fine, just still not CAPS LOCK.

gardners, 2020-04-03:
Okay, that has it so that we have a simple bitstream that takes only about 6 minutes to build, and that echos all typed keys out on the serial port, and that reproduces the problem.

It also confirms that CAPS LOCK is not showing up in the bucky bit vector.

gardners, 2020-04-03:
Above commits get CAPS LOCK working and seem to have also fixed the dropping of rapid key presses. Cursor up/down and left/right are confirmed to still be working in the hdmidebug test stream as well.  Now synthesising mega65r2 target to test the corrections.

gardners, 2020-04-03:
It worked, except that CAPS LOCK was inverted for the ASCII key parser, so have inverted the static unused widget board caps lock line for the MEGA65 R2 target to flip it.

gardners, 2020-04-03:
ok. All done and fixed, ready for testing and merging in.

adtbm, 2020-04-03:
Test was successfull.
cursor keys do work in all menus and normal operations as expected.
(cursor up and left, no problems anymore.)
CAPS LOCK works in all menus, especially FDISC.
(of course not in c64 mode, that's expected)

frehwagen, 2020-04-09:
Fixed as part of #156, merged into stable_base/14.

closed 2020-04-09 by frehwagen