MEGA65/mega65-core #190: Holding TAB on power-on not starting flash menu

gardners, 2020-03-28:
This seems to be a regression on unstable after re-basing onto the updates on stable_base. The problem also exhibits in the #156 feature branch.

Tackling the problem on unstable first. Suspecting that HICKUP.M65 was out of date.  Confirmed that latest HICKUP.M65 I have built DOES start flashmenu correctly. Now synthesising a bitstream with it incorporated to verify.

gardners, 2020-03-29:
It looks like HYPPO starts the flash menu, but the flash menu doesn't bring up the menu for some reason. Investigating...

gardners, 2020-03-29:
Looks like HYPPO now boots so fast, that it is unlikely that a key event will be triggered in the VERY short time at the beginning of HYPPO initialisation.

Solutions include adding a delay, or using a bucky key, like NO SCROLL instead of TAB, as the bucky keys are scanned instantaneously.  For some reason, we can't use $D613 to query the 9th column of keys, so we can't just hard-probe the TAB key that way.  We could, ,however, do a proper column 8 scan of it, but that will eat more code space that we would probably like.

adtbm, 2020-04-07:
NO SCROLL works as hotkey to enter flash menu. anyway, holding TAB while powering on
still brings a popup screen. this is not needed anymore

frehwagen, 2020-04-09:
Fixed as part of #156, merged stable_base/14.

closed 2020-04-09 by frehwagen