MEGA65/mega65-core #191: too many empty lines in hypervisor durinf bootup

adtbm, 2020-03-29:
during bootup there are too many empty lines in the Hypervisors info display, between
DISC-COUNT and
MEGA65.D81 image....... (I think it's 3 empty lines)

due to this, some of the information is scrolling out of the display.

Also the Bitstream version is not visible, when holding CTRL during bootup (probably due to this)

adtbm, 2020-03-30:
[image]

[image]
[image]

rdpeake, 2021-02-28:
[image]

current ctrl boot screen no longer has empty lines and shows a fair bit more details

closed 2021-02-28 by rdpeake