MEGA65/mega65-core #192: Revise virtual F011 behaviour

frehwagen, 2020-03-30:
Currently the virtual F011 mode has the following properties:

- is usable for all non-real-floppy drives
- is enabled for drive 0 and drive 1 (in sd mode) at once
- is under hypervisor control (accessible via serial monitor)

For more easy integration into apps and more flexibilty for development I propose the following changes:

- enable the virtual mode per F011 drive (independent if it is sd card or real floppy)
- enable control/status access from app side, not privilaged to hypervisor (for instance to enable apps to understand if mounting is possible or not)

gardners, 2020-03-30:
The Hypervisor should be the only thing that messes with the mounting of disk images. User should be allowed to change setting between SD card, real drive and virtual, however, and to see which mode a drive is in.  Important is that only the Hypervisor be allowed to write to the SD sector number for the drives. This will be enforced in the future.

frehwagen, 2020-03-31:
So I'll not be able to change mounted disk from within GEOS? Then I need to stop my work on "GEOS MOUNT" tool. I think especially for GEOS this is a very important functionality.

frehwagen, 2020-03-31:
Ah, ok I have to use the Hypervisor trap to change the mount, thats ok. Changing etting between SD card, real drive and virtual, an reading the status is then via direct register access or via hypervisor call as well?

gardners, 2020-03-31:
Hello,

Changing the source should still be fine from a register, as there is no
real security issue there, except when you want to change it back to SD
card.  So assume to attach and SD card disk image -- even if the last one
used -- that you will need to ask the hypervisor to do it.

Paul.

frehwagen, 2020-04-02:
Ok, lets try this change (so also covers #193):

Proposal:

Only changeable when in hypervisor mode:
$D68C-$D68F - F011 disk 1 disk image address on SD card
$D690-$D693 - F011 disk 2 disk image address on SD card

Set to one is only possible when in hypervisor mode:
GS $D6A1.0 F011:DRV0EN Use real floppy drive instead of SD card for 1st floppy drive
GS $D6A1.0 F011:DRV2EN Use real floppy drive instead of SD card for 2nd floppy drive

Deprecate (it is used for both f011 drive when in sd mode):
GS $D659.0 HCPU:VFLOP 1=Virtualise SD/Floppy access (usually for access via serial debugger interface)

Use those flags to enable f011 virtual mode separatly for the drive
$D689.5 VFLOP0 1=Virtualise SD/Floppy access for drive 0
$D689.6 VFLOP1 1=Virtualise SD/Floppy access for drive 0

gardners, 2020-04-02:
That looks fine to me. We then have to also patch monitor_load as well.
I am actually thinking about making it easier for monitor_load to access these protected registers, by having the hypervisor mode flag be asserted when the CPU does monitor memory accesses.  More precisely, have a privileged_access flag that is set when either the CPU is in hypervisor mode, or doing a monitor memory access.

frehwagen, 2020-04-13:
Base on the initial implementation of this feature branch I advanced as following:

1.) Enable the following flags as Read-Only:
Use those flags to enable f011 virtual mode separatly for the drives
$D689.5 VFLOP0 1=Virtualise SD/Floppy access for drive 0
$D689.6 VFLOP1 1=Virtualise SD/Floppy access for drive 1
Because the application (Non-Hyppo) need to understand this because if this is enabed mounting of SD D81 images as well as enabling real floppy will not be possible then.

2.) Let virtual F011 override read and write of F011 for drive 0 and 1.
So far the virtual F011 bit only applied to SD/D81 and to floppy write.

gardners, 2020-04-17:
Have looked through the code and accept the changes.

frehwagen, 2020-04-25:
I combined this 192- feature branch with the changes for #207 and #165 for testing together.

frehwagen, 2020-05-23:
I will add more branches for final testing here: #220, #217

frehwagen, 2020-05-24:
With #217 also includes #147. #220 already has been merged into stable_base for #200, so not relevant here anymore.

frehwagen, 2020-06-20:
I integrate #129 here, so I have to update my proposal because some of the registers/bits cause duplicate use.

Updated proposal:

Only changeable when in hypervisor mode:
$D68C-$D68F - F011 disk 1 disk image address on SD card
$D690-$D693 - F011 disk 2 disk image address on SD card

$D659.0 HCPU:VFLOP 1=Virtualise SD/Floppy0 access (usually for access via serial debugger interface)
$D659.1 HCPU:VFLOP 1=Virtualise SD/Floppy1 access (usually for access via serial debugger interface)

; control swap drive 0/1 (from #217 but updated)
$D689.5 - F011 swap drive 0 / 1

Set to zero is only possible when in hypervisor mode (via SD mount):
$D6A1.0 F011:DRV0EN Use real floppy drive instead of SD card for 1st floppy drive
$D6A1.2 F011:DRV2EN Use real floppy drive instead of SD card for 2nd floppy drive

Use those flags to read status of virtual mode separatly for the drives (from application)
$D68A.2 VFLOP0 1=Virtualise SD/Floppy access for drive 0 [R/O]
$D68A.3 VFLOP1 1=Virtualise SD/Floppy access for drive 1 [R/O]

gardners, 2020-06-21:
Looks fine to me. Let me know when you have it ready, and I'll update 138-hdmi-audio-27mhz to have these changes in.

lydon42, 2022-10-02:
virtual f011 does work, sometime bugginess I attribute to poor serial handling in m65. So until the implementation there is more clear, I close this feature request.

closed 2022-10-02 by lydon42