MEGA65/mega65-core #196: Support for German and selected other international/weird symbols

gardners, 2020-04-04:
It would be nice to be able to type German special letters etc, since ~1/3 the interest for the MEGA65 comes from German-speaking countries.  

Implementing much of the Unicode Latin1 page via chords with the ALT key (and ALT+shift, where it makes sense) seems the easiest way to do this.

gardners, 2020-04-04:
This will also require implementing the missing 128 symbols for our ASCII type face.

gardners, 2020-04-04:
NOTE: It is the DOS-style Latin1 codes that are generated, not UTF8-compatible codes that get generated.  This is unlikely to change.

gardners, 2020-04-08:
We have the first Latin1 supporting font in FDISK now, so we can begin to test this.
The keyboard layout is all over the shop, and CAPS LOCK has inverted meaning, and shift is ignored, so there is some fixing to be done there...

frehwagen, 2020-04-09:
Fixed as part of #156, merged into stable_base/14.

closed 2020-04-09 by frehwagen