MEGA65/mega65-core #197: ESC, F9, F11, F13 and TAB keys in ASCII key scanner have no delay before repeating

gardners, 2020-04-04:
Try pressing any of these keys in FDISK when prompted to type, and you will see this problem.
All are suspiciously on the last column of the keyboard scanner.

On-screen keyboard shows these keys as not being solidly on, so the problem is very much real and observable. 

Keys look to be held about 1/4 of the time.

##BREAKS ESC, F9, F11, F13 and TAB keys repeat at normal rate in ASCII keyboard scanner

adtbm, 2020-04-07:
Not sure, if this is related.
when pressing the above mentioned keys, wrong & strange symbols showing up.
Also when usuing standard keys, it looks like the letters are slowly shifting from left to right.
Have a look at the attached picture.

adtbm, 2020-04-07:
[image]

frehwagen, 2020-04-09:
Fixed as part of #156, merged into stable_base/14.

closed 2020-04-09 by frehwagen