MEGA65/mega65-core #199: when typing letters, they are getting displayed "shifted"

adtbm, 2020-04-07:
when repeatedly typing the same letter, it looks like, the letter is getting displayed slightly shifted every time.

please check this picture:
[image]

gardners, 2020-04-07:
Which bitstream? PAL/NTSC? VGA/HDMI?

gardners, 2020-04-07:
If it is the #156 fix bitstream, then I wouldn't worry about it, as it will get totally re-worked when the revised HDMI mode stuff gets pulled in.

frehwagen, 2020-04-07:
Do you know, can you explain what exact change fixes this from the changes we consider HDMI rework?

gardners, 2020-04-07:
The problem shown above is due to the old display mode stuff. It should completely go away with the HDMI rework, so I think we can just close this issue as "fixed by unmerged changes", and only worry about it if it reappears.  If you agree, just close it.

adtbm, 2020-04-08:
Rechecked the issue.
it is bound to the general HDMI/VGA rework requirement.

The bug only appears in VGA NTSC all others are fine

lydon42, 2021-12-21:
I can't see this on nexysA7 VGA, neither with master, nor development/hwsc. Checked PAL/NTSC, CRT emulation on/off, everything looks fine.
Closing.

closed 2021-12-21 by lydon42