MEGA65/mega65-core #200: PAL/NTSC default setting in config menu is being ignored

deftmega, 2020-04-08:
In latest unstable bitstreams (sorry for no further info at this point) I have no functionality by the PAL/NTSC toggle for default video mode in config menu. However, switching via "V" key in freeze menu has been tested working.

adtbm, 2020-05-23:
The Bug is still in.

gardners, 2020-05-23:
The problem is that mega65 configure is modifying sector 1 of the SD card, while the hypervisor reads from sector 1 of the system partition.  Both use the hypervisor routine to do the reading and writing, so something funny is going on with that, I presume.

gardners, 2020-05-23:
ok. Found problem: Util menu check happens before SYSPART can be opened. Thus syspart offset  is sector $0, and the config data thus ends up in sector 0+1 of the SD card.  This is actually not a bad place for it, and the easier fix is probably to just make it officially live there.

gardners, 2020-05-23:
Confirmed fixed by the above commit.

It's on the 138-hdmi-audio-27mhz branch, and will get merged in when that branch gets merged in.

adtbm, 2020-05-23:
Tested Bitstream:

bluewaysw, 2020-05-24:
Combined with #220, successfully tested and merged into stable_base branch: https://github.com/MEGA65/mega65-core/commit/0476b7475b6ddc84380ca12ec10a8fb7fea04174

closed 2020-05-24 by frehwagen