MEGA65/mega65-core #202: Machine locks up with cartridge inserted

deftmega, 2020-04-09:
Encounterd in 156/65: Machine won't boot if cartridge is inserted. Also not possible to run flash or any other menu. Blue screen with interesting white block visible.

deftmega, 2020-10-12:
This is still the case for easyflash. Behavior changes on each reset (reproducible) until menu appears. Can't start any title tho. Other cartridges e.g. Sam's Journey work.

deftmega, 2022-01-03:
easyflash 2 is no longer working. this is still the case. Sam's Journey and super breadbox work. We will look into this later.

lydon42, 2022-05-29:
this is a #564

lydon42, 2023-04-10:
Ultimax boot works with [3fcf2de](https://github.com/MEGA65/mega65-core/commit/3fcf2deceaada30f1a8e1b219eb943e737211d65)
This is a duplicate of #684

closed 2023-04-10 by lydon42