MEGA65/mega65-core #204: Double speed of HyperRAM interface from 40MHz to 80MHz

gardners, 2020-04-11:
We need 4x clock for our state machine, so this means we need a 325MHz clock to drive what will must be a very simple state machine because of the high clock speed.

Alternatively, we can use the 325MHz clock to feed a pair of ISERDES and OSERDES unit to do the high-speed stuff.

This is needed bot to improve the HyperRAM performance in the MEGA65 core, but also as a pre-requisite for the Amiga core that may well require lower latency than the 40MHz clock can deliver.

lydon42, 2024-08-17:
This is a very old branch, and it seems not to have been merged. But I would assume that, after all the HyperRAM and SDRAM changes, it will be kind of hard to still use this.

Do you agree @gardners @frehwagen ? Please close if yes and tell me if I should kill the branch.

gardners, 2024-08-17:
I agree, we leave HyperRAM as is now.  I would be happy for us to close the issue @lydon42

closed 2024-08-17 by lydon42