MEGA65/mega65-core #207: Some hyperviser call destroy RAM at $CE00-$CEFF

frehwagen, 2020-04-14:
When using the hypervisor traps to control SD/DOS some of the function unexpectly destroy $CE00-$CEFF which is a bit of an hidden contract and not very nice to handle for some application as GEOS, because there is always the need to save the memory first and restore it afterwards.

We should try to prevent those memory corruption outside of the HYPPO area $4000-$8000 when using hypervisor traps.

gardners, 2020-04-14:
I completely agree.  I think this snuck in during the conversion to KickAss, and when there was no space left in HYPPO.  But a number of things have been optimised or removed as no longer required, and there is no need for this to happen anymore.

Using the HyperRAM mapped at $4000-$7FFF is an option, but not all MEGA65 targets have expansion RAM, so I intend to just remove the data segment entirely.

gardners, 2020-04-15:
See above commit on new 207-no-ce00 branch.
Please test

frehwagen, 2020-05-24:
Cherry picket into #192 branch for testing together.

frehwagen, 2020-05-31:
Merged to stable_base as part of #222 into release @stable_base/16.

closed 2020-05-31 by frehwagen