MEGA65/mega65-core #208: Implement keyboard test program for manufacture

gardners, 2020-04-16:
GMK need a way to easily test the keyboads they manufacture.
The plan is to have a program that shows the virtual keyboard, and blinks the keyboard LEDs in different colours.  
This will be launched from the utility menu, and will allow users to confirm whether their own keyboard has developed a fault as well.

gardners, 2020-04-16:
LED blinking will be done by pressing keys: Each key press registered will cycle the colours through, with each of the four LEDs going through R, G, B in turn, so that exactly one colour channel is active at any point in time.

lydon42, 2022-01-04:
Is the result of this the KEYBOARD TEST in the ALT menu? I fail to see how to test LEDs with this.

closed 2024-08-17 by lydon42