MEGA65/mega65-core #212: Accessing external drive should not trigger sounds from speaker

RalphEgas, 2020-04-26:
Context:

By default device #9 is set to an external drive. I tried addressing an external drive by executing load"$",9. I didn't actually have an external drive hooked up.

Witnessed behavior:

When I issues the command, sound was audible from the internal speaker that sounded like a motor was spinning up and coming to a close in rapid succession.

Expected behaviour:

No sound should be audible for external drive access.

gardners, 2020-04-27:
As we discussed, your machine has no internal speaker :) You are hearing the real floppy drive tracking the head movements that the SD card would have triggered, had they been real disk accesses.  Thus this is expected and correct behaviour

RalphEgas, 2020-04-27:
@gardners well, I'm not convinced that this issue can be dismissed, regardless of the sound coming from an actual internal drive. The issue as stated above addresses the fact that an internal sound is heard from an external drive that wasn't even hooked up.

In the external drive's absence, no sound should be heard when accessing one.

What would happen if one actually ended up hooking up an external drive. Would you then hear both external and internal drive spin up their motors? That can't be right.

gardners, 2020-04-27:
The noise from the internal drive only occurs when accessing a disk image from the SD card, ie., there is no real drive as the data source. When a real drive is the data source, there is no "ghosting" by the internal drive, just the real drive actions, so you can't get duplicated noise.  But as to whether we want the external drive over SD to make the internal drive whirr, that probably comes down to a user experience thing. My feeling is that it would be weirder to have ,8 make noise, but not ,9, even though both are coming from SD card.  Probably worth checking with Deft for a final decision.

RalphEgas, 2020-04-27:
I think you're absolutely right with regards to consistency in hearing sounds for #8 and #9 if both come from an SD card. What I'm arguing though is sound coming off an external drive that is *not* tied to an SD card. But then again, my assumption that I was selecting an actual external drive (not and SD card equivalent) may have been wrong and you already assumed this to be the case may have given rise to your argument. To be clear, I selected 1565 drive 1 under (1) External 1565 and assume this would access an external drive hooked up to the serial port. If that is not what this setting means, but instead it still means I'm somehow hooking up #9 to an internal thing, then I understand where you're coming from. Then still, the wording on the device selection in the freeze menu is hopelessly non-intuitive.

frehwagen, 2020-04-27:
The internal logic is real floppy internal or external map ot the same physical port at the moment, so you acutually have an external one it is the internal.

The source on sd mounts is a different issue.

frehwagen, 2020-04-27:
The issue is more that freezer offer the external drive that is not supported at the moment, I would think.

gardners, 2020-04-28:
Well, the problem is that the F011 doesn't correctly activate the external drive when selected. It is presumably still driving the same select line. The hardware *should* with a dual 34-pin floppy cable work with 2 drives.  This is of course untested at the moment.  

(Note that the 2nd drive on a finished machine, is intended to be via a 1565 compatible C65 external floppy connection, not a normal C64 serial connection. Or people can just hang the 34-pin floppy cable and power out the back if they want an ugly solution.)

Ok. I just looked into this, and the 2nd floppy drive is not yet plumbed through, and does indeed use the internal drive mechanism.  I'm about to file an issue for this, and submit a fix for it.

gardners, 2020-04-28:
Ok. Synthesising that fix.  This will mean trying to use floppy drive 2 with a single-drive MEGA65 will fail, instead of using the internal mechanism. I don't have a 3.5" drive or disks here, but I do have 5.25" ones, somewhat ironically.

frehwagen, 2020-04-28:
Reopend because we are doing some code changes here.

But isn't it that we now loose the default option to have mounted drive 0/8 (SD) and use internal floppy as drive 1/9? GEOS implementation at least relies the fact the drive 1 floppy is actually map to the internal drive. 

I think we need an option to map between F011 and internal floppy drive number more flexible.

frehwagen, 2020-04-28:
Ah it is done on #217.

closed 2020-04-28 by frehwagen