MEGA65/mega65-core #214: Make settings in freezer menu persistent

RalphEgas, 2020-04-26:
In the freezer menu it's possible to configure drive mapping and a few other things. Tweaking these values take time but all those tweak get reset to their default values upon soft and hard resets. To have this not happen would be a definite QoL improvement.

This is what Paul commented on it:
@ralph: not yet. but theoretically possible to do.  We would actually implement it by having the audio mixer saving them to SD card or NVRAM, and then the flash menu on startup when it does the bitstream checks to apply the saved coefficients.

gardners, 2020-04-27:
I just edited the title: In this issue we will only deal with making the audio settings saveable. Floppy settings should be done via MEGA65 configure programme, where in fact, most of them can already be done.

gardners, 2020-04-27:
We can either have the audio settings on an SD card sector, or in NVRAM of RTC. Yet to be decided.

lydon42, 2022-01-05:
I would opt for saving audio settings to RTC. This way you always have a sane setting for your environment when starting up.
Perhaps we could save audio settings in addition to that also to a freeze slot? So that the settings of a freeze a kept if a certain software needs special settings?

WebbedQuasar, 2024-10-30:
My recently received Mega65 with the R6 board significantly overdrives the volume at the default (0dB Master Volume). I recorded a short video with my phone to show how my M65 sounds right after turning it on, where everything should actually be silent on the Basic screen: https://youtu.be/o1eCfImNaF4
The only way to eliminate this annoying noise (which is, at default, still audible during the brief loading pauses when playing Showdown for example, so it continuously taps in the background like a background noise) is to reduce the Master Volume on both channels by at least -5dB. Sometimes, I have to set it to -9dB to make the noise disappear, and then increase my TV's volume to around 50 out of 100 for normal volume level. It’s inconvenient to have to adjust this again after every reset or reboot. It wouldn’t necessarily require saving all settings of the Audio Mixer, just the Master Volume. Perhaps there could be an option to save the Master Volume in the Config menu?
Additionally, I learned on Forum64 that not all Mega65 units are affected, only some. Here’s my thread on the issue (in German): https://www.forum64.de/index.php?thread/151318-st%C3%B6rger%C3%A4usche-im-mega65-core-auf-hdmi-with-sound
It’s definitely a peculiarity of the Mega65 core, as this noise doesn’t occur in the C64 core where everything is quiet after booting. If there were a way to fix this strange noise directly in the core, all the better.