MEGA65/mega65-core #215: SDHC cards of 16GB or larger don't like ACMD41 initialisation

gardners, 2020-04-27:
SDHC cards don't like the ACMD41 initialisation sequence, as it makes them think the host is SDC only, and so they allow access to only a small fraction of the storage. (See https://electronics.stackexchange.com/questions/358114/sd-card-block-read-write-issue-in-spi-mode-stm32-cmd17-cmd24 )

So will try to fix this.

##BREAKS 16GB SDHC card capacity correctly recognised in FDISK

gardners, 2020-04-27:
Note this seems to mostly be a problem with the external microSD slot. 8GB and 16GB cards work on the internal slot for me, but neither properly on the external slot.

gardners, 2020-04-27:
Seems to work, but there was actually a problem in FDISK that was also responsible.

lydon42, 2022-01-05:
Tested with latest prerelease devel@fc504ef and 920283: 16GB µSDHC is correctly detected and used in external slot.

closed 2022-01-05 by lydon42