MEGA65/mega65-core #217: Selecting second real floppy drive has no effect

gardners, 2020-04-28:
Currently we don't control the 2nd floppy drive, but still drive the internal drive.  This needs to be fixed to control a 2nd drive on the floppy cable (like B: drive on PCs).

Triggered by the discussion at #212

gardners, 2020-04-28:
Created 217-secondfloppy feature branch.
Synthesising now. We will need to test that internal drive still works (I have no 3.5" floppies here), and then try a 2nd floppy drive.

gardners, 2020-04-28:
Seeking track 0 still hangs forever, because the C65 ROM doesn't seem to have a mechanism for limiting the number of track steps that it will attempt before giving up.  This may be because the F011 external drives show low values by default, if not real drive is connected.

frehwagen, 2020-04-28:
Repeat from  #212 

Reopend because we are doing some code changes here.

But isn't it that we now loose the default option to have mounted drive 0/8 (SD) and use internal floppy as drive 1/9? GEOS implementation at least relies the fact the drive 1 floppy is actually map to the internal drive.

I think we need an option to map between F011 and internal floppy drive number more flexible.

gardners, 2020-04-28:
Yes, we can add a flag to swap the internal and external floppy IDs, so that we have the best of all worlds.

RalphEgas, 2020-04-28:
@gardners I noticed an infinite hang until I actually put in a 3.5" floppy into the internal real drive as well, btw. I don't know whether this is standard behaviour for a F011. I don't think this was the case of 1581 drives on the C64 though. That would just report an error immediately if no floppy was inserted.

gardners, 2020-04-28:
That's fixed in the unstable branch, but has yet to be merged in.

Paul.

gardners, 2020-04-28:
ok. I am working on allowing swapping internal/external drive for @frehwagen, and also including the "no infinite hang" problem for you @RalphEgas 
Will share bitstream when cooked.

frehwagen, 2020-05-24:
Cherry picked to #192 branch, for testing together.

frehwagen, 2020-05-24:
Includes #147

lydon42, 2024-08-17:
@gardners do you remember if this is finished? The tagged commits are all in release-0.96 and development.