MEGA65/mega65-core #218: Error while Compiling

Nato-Hendo, 2020-05-03:
I'm currently attempting to compile via make and am receiving errors from Vivado stating the Synthesis failed. I've attached a text file containing the console output detailing the error(s). Everything instructed in the 'build' documentation file has been followed so I'm not sure what I'm doing wrong, thanks.

[compiling_error.txt](https://github.com/MEGA65/mega65-core/files/4569660/compiling_error.txt)

gardners, 2020-05-03:
Hello,

use:

make bin/mega65r2.bit

and preferably on the stable_base branch.

We need to clean up the default target and a few other bits and pieces, so
sorry about the confusion.
What board are you targeting?

Let me know if you hit any other problems.

Paul

Nato-Hendo, 2020-05-05:
Hi Paul,

Thanks for the assistance, I've followed your advice but am now encountering another error which I assume relates to these error messages as they appear just prior to the failure:

INFO: [Synth 8-139] cannot merge instances iomapper0i_10/\block2.framepacker0/buffer_moby_toggle_reg  and iomapper0i_10/\block2.framepacker0/buffer_offset_reg[11]  because of non-equivalent assertions
INFO: [Synth 8-139] cannot merge instances iomapper0i_10/\block2.framepacker0/buffer_moby_toggle_reg  and iomapper0i_10/\block2.framepacker0/buffer_offset_reg[11]  because of non-equivalent assertions

I've attached a snippet of the output from the terminal.

[compiling_error_02.txt](https://github.com/MEGA65/mega65-core/files/4578514/compiling_error_02.txt)

Thanks, Nathan.

gardners, 2020-05-05:
Hello,

That's weird.
Which branch and commit?

Paul.

Nato-Hendo, 2020-05-05:
According to git show:
commit 4abef64aee57c16b88e2cfa144bf17c936755c55 (HEAD -> stable_base, origin/stable_base)

gardners, 2020-05-05:
> According to git show:
> commit 4abef64
> (HEAD -> stable_base, origin/stable_base)

Can you please try:

git checkout 165-hyperram
make bin/mega65r2.bit

And tell me how that goes?

Thanks,
Paul.

Nato-Hendo, 2020-05-06:
Using those, I receive the following error:

make: *** No rule to make target 'src/vhdl/gen_utils.vhd', needed by 'vivado/mega65r2.xpr'.  Stop.

[compiling_error_03.txt](https://github.com/MEGA65/mega65-core/files/4584743/compiling_error_03.txt)

gardners, 2020-05-07:
Hello,

Ok. I have just added and pushed those.  Let me know if it still doesn't
build. But if you try in the next couple of hours, do another pull and try
again, as I am actively working on a couple of things for that branch.

Paul.

Nato-Hendo, 2020-05-07:
Thanks Paul, after pulling I received this error:

make: *** No rule to make target 'src/vhdl/s27kl0641.vhd', needed by 'vivado/mega65r2.xpr'.  Stop.

gardners, 2020-05-07:
Hello,

Hmm.. ok. That's one that is only used for simulation.  Pushing a fix to
the Makefile for this.

Paul.

rdpeake, 2021-02-28:
closing due to inactivity - builds via development branch should be working at this time for r2 board though

closed 2021-02-28 by rdpeake