MEGA65/mega65-core #220: Making CC65 tools should be optional in the make process

bluewaysw, 2020-05-16:
To optimize build times (especially for CI) building CC65 should be optional.

The idea is to pass an parameter to the make to force the build process to use a local installed cc65 version. Like:

make USE_LOCAL_CC65=1

We need to pass this behaviour to all submodules.

bluewaysw, 2020-05-24:
Fixed and sent to stable_base branch together with #200: https://github.com/MEGA65/mega65-core/commit/0476b7475b6ddc84380ca12ec10a8fb7fea04174

closed 2020-05-24 by bluewaysw

Ben-401, 2020-06-15:
this feature does not work for me.

from a clean checkout, and using plain old "make" will actually use the local "cc65"
which I dont have installed.
It should infact init/update/build the "cc65" SUBMODULE, and set the paths to point to the SUBMODULE and then set the path(s) to the SUBMODULE.

When using ```make USE_LOCAL_CC65=1``` does set the path to using LOCAL "cc65"
but logic seems incorrect, ie instead of:
```ifeq ($(USE_LOCAL_CC65),"")``` (in Makefile)
i suggest to use:
```ifeq ($(USE_LOCAL_CC65),"1")```
so that the USE_LOCAL_CC65==1 is only asserted when passed to "make" during the build.

And the default "make" behavior should be to init/update/build SUBMODULE to allow for standalone and self sufficient building.

I have a patch for this in stable_base, but before I go ahead, it seems the other submodules also require updating to this logic, namely:
mega65-freezemenu and mega65-fdisk

Does this seem fair? @bluewaysw @frehwagen @gardners

frehwagen, 2020-06-15:
Ok, so the "" check fails if the USE_LOCAL_CC65 is not specified at all? This sound reasonable. Basically it is building CC65 in that case (what I double checked) but not uses it, right?

Ben-401, 2020-06-15:
Hi, yes the "" fails (causing LOCAL paths to be used) even when not specified on the CLI.
Actually the building also does not get executed either due to the logic used.