MEGA65/mega65-core #222: Allow querying and setting of real time clock from MEGA65 Configure

gardners, 2020-05-26:


gardners, 2020-05-26:
Implemented. Should be tested for consideration for merging onto stable_base.

gardners, 2020-05-26:
As suggested improvement is to always use 24 hour time, and hide bit 7 of the hour, to make it less confusing for users. Probably not a bad suggestion.

adtbm, 2020-05-28:
Tested #222 - 1
Bitstream used: mega65r2-20200527.10-HEAD_n-a9ccc62+DIRTY.bit

Main Test - RTC clock
- Setting the RTC clock works
nicely implemented in the Settings menu. 24hr time format

Things noticed:
- Defts Demo crashes/freezes MEGA65, see pictures

- PAL
[image]

-NTSC
[image]

Using 165-88 as reference: Demo works.

gardners, 2020-05-28:
The problem is some VIC-IV issues that have already been fixed in #165 branch. i.e., it is not a blocking problem that the demo doesn't work.

frehwagen, 2020-05-31:
Merged into stable_base for release @stable_base/16, this includes #207 fix as well.

closed 2020-05-31 by frehwagen