MEGA65/mega65-core #224: python2: No such file or directory

AidenJ-jaco0160, 2020-05-27:
I received a compilation error while attempting to compile using the Makefile, the output can be seen below

Makefile:425: ~~~~~~~~~~~~~~~~> Making: src/utilities/ethertest.prg
Ophis/bin/ophis -4 src/utilities/ethertest.a65 -l src/utilities/ethertest.list -m src/utilities/ethertest.map -o src/utilities/ethertest.prg
/usr/bin/env: ‘python2’: No such file or directory
Makefile:425: recipe for target 'src/utilities/ethertest.prg' failed
make: *** [src/utilities/ethertest.prg] Error 127

Is python2 another file or piece of software that I need to install? Cheers,
- Aiden J

gardners, 2020-05-27:
python2 is version 2 of the python programming language. You should be able to install this fairly easily. If on a Linux-y system, try something like:

`sudo apt-get install python2
`

Ben-401, 2020-06-24:
can be closed, as we have "python2" listed as a dependancy.
Just need to get stable_base onto the master-branch for newbies to more easily see the dependencies..

closed 2021-02-28 by rdpeake