MEGA65/mega65-core #225: Insert RTC via "hyperrupt" into TI$ at MEGA65 startup

adtbm, 2020-05-27:
It would be great, if during powerup, the MEGA65s RTC values could be once inserted into TI$.

lgblgblgb, 2020-06-21:
just my random thoughts here, with the hope that can be useful for something at least:

Btw, AFAIK C65 reads TOD of CIA1 (or 2?) as TI$. At least when I wrote CIA emulation on Xemu, just having the emulation to get back "real time" when TOD is read caused TI$ become correct ... What I mean here, that this can be rephrased as putting real-time from RTC to TOD(s?) of CIA(s)? Also if some don't write TOD at all, basically reading TOD can just mean reading the RTC :) Though writing TOD registers should not mean writing RTC, it can cause silly problems if a software modify TOD for whatever reason. But for the "boot-up default" it makes sense (IMHO) to have the same time for TOD & RTC (it's another question if they can diverge because the different clocking, etc?).

lydon42, 2022-09-25:
I think this is antique and can be closed.

closed 2022-09-25 by lydon42