MEGA65/mega65-core #227: 6502 Illegal Opcode Support

gardners, 2020-05-31:
Currently we support the 4510/45GS02 instruction set the whole time.
But some C64 software uses 6502 illegal opcodes. We should support this.
There is already provisional support via the force_4502 flag.

Now that we have all the illegal opcodes documented in the MEGA65 Book, we can work to implementing them to that spec.

gardners, 2020-05-31:
There is already a HyperTrap call to toggle it:

```
LDA #$72
STA $D620
NOP
```

Freeze Menu also supports C to toggle the mode, and shows it.

Thus we have all the ingredients already there, and we just need to examine the support for the opcodes in the CPU, and check cycle timing etc.

gardners, 2020-05-31:
Wrote a little test programme that confirms if we are in 6502 or 4510 mode:

```
LDA #$00
STA $FD
LDA #$FF 
TSB $FD
```

In 6502 mode, TSB becomes a NOP, and $FD will contain $00. But in 4510 mode, TSB will copy the set bits from A, and $FD will contain $FF.

Thus we just need to go through and implement all the illegal opcode instructions.

gardners, 2021-02-07:
Above commit tries to fix C65 DOS going bonkers, as it was not in $E000-$FFFF ROM area, so was being run in 6502 mode, instead of 4510 mode.  Now we check if a MAP instruction has been used to map memory anywhere. If so, then we stay in 4510 mode.

gardners, 2021-02-07:
... but doesn't work.
To test:
1. Boot to C64 mode
2. Use freeze menu to select "auto" CPU mode
3. SYS 64738 to force DOS to reset
4. LOAD"$ on a disk image with a decently long directory
5. Observe that directory is corrupted

My gut feeling is that the CPU is staying in 6502 mode at some point, when it should be in 4510 mode for the DOS code... But it needs more research to confirm it.

deftmega, 2022-01-03:
we deccided against illegal opcode. Get the C64 core if you love them.

closed 2022-01-04 by lydon42