MEGA65/mega65-core #229: Error while building SDL2-2.0.5

cond0068, 2020-06-09:
While installing dependencies required for the [installation of the xemu](https://github.com/MEGA65/mega65-core/blob/master/docs/xemu.md),
i received this error and i could not build the SDL2-2.0.5 library that was mentioned.
```
build/.libs/SDL_mirvideo.o: In function `MIR_InitDisplayFromOutput':
> /home/student/unit4/SDL2-2.0.5/src/video/mir/SDL_mirvideo.c:258: undefined reference to `mir_output_get_current_mode'
> collect2: error: ld returned 1 exit status
> Makefile:147: recipe for target 'build/libSDL2.la' failed
> make: *** [build/libSDL2.la] Error 1
```
To avoid this error, step number 6 for the installation of SDL2 libraries should have "--enable-mir-shared=no" appended to it as it will link the mir library sucessfully and remove the error. 
```
To install the SDL2 libraries:

    in a browser, download the source at https://www.libsdl.org/release/SDL2-2.0.5.tar.gz this will be saved in ~/Downloads enter the following commands:
    1. cd ~/Downloads/
    2. tar xvfz SDL2-2.0.5.tar.gz
    3. cd SDL2-2.0.5/
    4. mkdir build
    5. cd build
    6. ../configure
    7. grep MISSING config.status NOTE: see below regarding the above command, ie re VM
    8. make
    9. sudo make install
```
```
    6. ../configure --enable-mir-shared=no
```

Alternatively, the variable "EXTRA_LDFLAGS" within the Makefile could have "-lmirclient" appended to it.

```
SDL2-2.0.5/build <Makefile>, "cat Makefile | grep EXTRA_LDFLAGS"
EXTRA_LDFLAGS =  -Wl,--no-undefined -lm -ldl -lsndio -lmirclient -lmircore -lEGL -lxkbcommon -lpthread -lrt
change to
EXTRA_LDFLAGS =  -Wl,--no-undefined -lm -ldl -lsndio -lmirclient -lmircore -lEGL -lxkbcommon -lpthread -lrt -lmirclient
``` 

I'm not sure if this will be useful but i thought i would share as a build of SDL2-2.0.5 is required to run the xemu.

# References
I do not take credit for this find it all goes to [stackoverflow](https://stackoverflow.com/questions/44758276/build-error-for-sdl2-2-0-5-in-ubuntu-16-04) with mentions of [the bug report](https://bugzilla.libsdl.org/show_bug.cgi?id=3539) & [the bug patch](https://hg.libsdl.org/SDL/rev/3034ea08d805).

lgblgblgb, 2020-06-09:
Thanks for the report. This is more like an SDL specific issue, not too much to do with Xemu or MEGA65 in general. The issue here is more about using a too old Linux distribution (in our case Ubuntu). Xemu needs at least SDL 2.0.4, which is the case on any decent enough distribution, so you don't even need to compile SDL yourself, just installing it with the package manager of your distribution (apt in case of Ubuntu, or in general any Debian derived distribution, ie `apt install libsdl2-dev` which also takes care dependencies and likes including xorg or any other packages). However with Ubuntu versions predates 18.04 it's possible that the SDL2 it would provide is simply too old, then indeed, you need. Personally I would recommend to use more up-to-date distribution, as for even LTS versions only, we have already 18.04 and 20.04.

Also please note, that in general, you should refer for the Xemu documentation more, rather than documentation inside mega65-core. https://github.com/lgblgblgb/xemu

You can also find some pre-built versions of Xemu: https://github.lgb.hu/xemu/

cond0068, 2020-06-09:
That's okay and yeah i thought it might be irrelevant towards the Xemu & MEGA65 project but hopefully it may just help someone. My class was specifically asked to use an outdated version of Ubuntu (18.04) - my guess, it was so we could have fun with issues like this. 

Thank you for the follow-up information @lgblgblgb i appreciate it!

lgblgblgb, 2020-06-09:
And if you need to compile SDL anyway, for whatever reason, I would opted for the latest SDL release, not 2.0.5 (I guess the document you followed was written as an example when 2.0.5 was the latest). It raises the question though, if it's really a good idea that we have documentations on the same topic (ie compiling Xemu in our case) at multiple places with some kind of divergent state, hmmm ...

cond0068, 2020-06-11:
Thank you! I have mentioned your recommendations to a fellow student that was having the same error.

closed 2020-06-11 by frehwagen