MEGA65/mega65-core #231: compulation error while running make

DianaBarber8880, 2020-06-10:
[image]
followed the directions in the file docs/build.md 
this error appears when I run the makefile with the make command

gardners, 2020-06-10:
The dead process MIGHT be because you don't have enough RAM -- expect Vivado to potentially peak above 8GB of RAM usage.  Similarly lack of disk space might cause trouble.  What's the spec of you system?

But excited to see other folks trying to build bitstreams!

gardners, 2020-06-10:
Yes, almost certainly the 4GB VM RAM is a problem.  What is your native OS? Vivado runs natively on Windows, for example, if you prefer to avoid the VM.

Ben-401, 2020-06-24:
my vm is allocated 8G and within the VM I can do the following:
- I can generate the bitstream using the Makefile,
- but when If loading the Vivado GUI, followed by Opening Implemented Design, the VM disappears from my Host- ie crashes.

Were you successful in building the design by allocating more RAM to the VM?

lydon42, 2021-12-20:
Not a bug, more something that should be handled on discord.
Also over an year old.
Closing.

closed 2021-12-20 by lydon42