MEGA65/mega65-core #233: Call for documentation revision

Kern0042, 2020-06-11:
## Brief
Whilst venturing down the lovely rabbit hole of getting the Megacore running, I believe that a lot of confusion whilst compiling could be cleared up with a revision of the documentation. <br /> <br />

## Suggestions 
### General Improvements
The index needs a clearer layout. This could be achieved by revising the naming conventions and blocking the documentation into sections, e.g. Compiling the MegaCore, Basic Interactions with the MegaCore and Advanced Interactions with the MegaCore.

### Getting Started/First time compiling experience for new members of the community.
Compiling the Megacore could included various basic guides to get started with specific version numbers and examples of what the user should be seeing. This is proposed primarily as a tool to show the member what they should expect so the member can get closer to understanding and contributing to the project. As it stands the documentation is there just disjointed and in need of some TLC.

### Basic Interactions 
May include a list and examples of the most basic interactions a user may be able to do with the program.

### Advanced Interactions
This may include a more collaborative approach allowing the community to contribute guides for how to use their latest features etc. hand in hand with the next suggestion.

### Function List
A list to document all the current features in the project and to tie all git hub commits to a central location such as index.

gardners, 2020-06-11:
It would be helpful if you made examples of what you are proposing.

Kern0042, 2020-06-11:
One complete guide could be all this project needs as long as all version numbers are listed so newbies could replicate the method if they were having issues for example compiling the dev build (mostly exposure so that when switching to a dev build they can undertstand whats happening). Each heading could link to its own page however if possible it would be amazing to have a full PDF using the structure mentioned, the current table of contents could resemble.

# Table of Contents:
1. Getting Started - How to successfully compile. A through guide using version XXXX of ubuntu 64bit methods
shown will work for version XXXX process may vary depending on distro of choice.
1.1 Requirements
1.2 Dependency - Version XXX
1.3 Method 
1.4 Example of a successful compile.
2. Basic Interactions - (Formally known as using.md) Various commands and functions and how to use the basic features.
2.1 Ect
2.2 ect
2.3 ect
3. Advanced Interactions - Community describes advanced commands ,debug tools or docs possibly discussion on tools to write a custom OS.
3.1 Tools -
3.2 Quirks of the Emulator
4. Helpful Links
4.1 MEGA65 - same as current table of contents but hyperlinked to its own page
4.2 C65
4.3 C64
4.4 Misc

Ben-401, 2020-06-12:
i agree that the documentation is out of date.

I agree that listing the updated requirements to build would be most useful. I just did a clean clone, master branch, "make", and it failed at about "iverilog" as autoconf was not found.

I understand the previous requirements were listed in:
https://github.com/MEGA65/mega65-core/blob/master/docs/build.md#compiling
but this is now outdated.

Ben-401, 2020-06-12:
i could at least provide a list of dependencies that my system complains about, and post them into a list to try and resolve what are the packages required. I use ubuntu 18.

Paul can you please advise what branch any updates to the build process should be applied to?
I would hope that the master branch file readme.md is at least updated as this is what people see either from a clean checkout/clone or via the github UI.

gardners, 2020-06-12:
Yes, that would be helpful.
master, stable_base and 138-hdmi-audio-27mhz are the three key branches at
the moment.
Basically stable_Base will get merged onto master at some point, and
138-hdmi-audio-27mhz onto stable.
If you can report the missing dependencies, that would be great.

Paul.

closed 2022-01-03 by lydon42