MEGA65/mega65-core #237: USE_LOCAL_CC65 just one cc65 submodule fdisk freezemenu

Ben-401, 2020-06-19:
a suggested improvement is:
When you dont have cc65 installed, and require cc65 to be built using the Makefile(s),
that only one copy of the cc65 submodule be built.

This suggestion affects submodules: mega65-fdisk and mega65-freezemenu.
and relates to #220.

Currently, three version of cc65 are built, one in the mega65-core and in the two submodules.

I suggest that the two submodules be able to detect if they are a submodule of mega65-core , and if so, to set the path to something like "../../../cc65/bin/cc65" so as to access the binary in the mega65-repo.

The benefits of this are just to reduce compile time (with clean checkout), reduced compile logs etc.

If this improvement is accepted, I welcome someone else to implement.

gardners, 2020-06-19:
The challenge here is that it is possible to build those sub-modules independently of the mega65-core repo, so they need to be able to stand alone. Now, when the mega65-core makefile calls into those sub-modules, it can pass the CC65 path to the make command line to achieve what you are proposing, while leaving the ability of those to be built independently.

Ben-401, 2020-06-19:
exactly,
mega65-core can pass say USE_PATH_CC65="../../../cc65/bin" and for standalone operation, each submodule first checks if USE_LOCAL_CC65 is set (to satisfy CI #220), and of not set, can check for the presence of this new USE_PATH_CC65 arg, and if set, to use that path. That will achieve this improvement.
If both USE_LOCAL_CC65 and USE_PATH_CC65 are not set, compile and use the cc65.

lydon42, 2022-01-03:
Actually this can be archived by passing CC65/CL65/CA65 with the make call to the submodule. For this the CX65 vars need to be cleaned up, as core sets them including options, but freezer for example expects only the binary and adds the options extra...

lydon42, 2022-01-03:
See #514

closed 2022-01-03 by lydon42