MEGA65/mega65-core #240: Open-roms Ophis not cloned on root make

rdpeake, 2020-06-21:
When attempting to run make on a fresh download of the repo, it fails to run the open-roms make file as it does not init/update the submodules in order to pull down Ophis. (manually doing this in the open-roms folder worked to get past that point.

Ben-401, 2020-06-21:
what branch are you using?
Please use stable_base branch and not master.
Please post the logs if there is an error.

rdpeake, 2020-06-21:
was using development as that was what was suggested by the build readme file - will try stable_base now

rdpeake, 2020-06-21:
Running on stable_base seemed to work up to vivaldo builds - though with the dependencies i'd already installed having to do it the first time.

Given i'm talking dependencies - i figure i should probably point out i needed to install the following in order to get make to run under windows subsystem for linux, ubuntu 18.04 LTS

make, gcc, libusb-1.0-0-dev, libpng-dev (not sure this is needed), autoconf, gperf, flex, bison, g++, openjdk-11-jre-headless, python-minimal, imagemagick

I also had to make sure the vivaldo i installed was available via /opt/ - as it installed into /tools/ by default.

The issue i have now is DRC: `ERROR: [DRC MDRV-1] Multiple Driver Nets: Net machine0/cpu0/reg_math_write has multiple drivers: machine0/cpu0/reg_math_write_reg/Q, and machine0/cpu0/reg_math_write_reg__0/Q.`

gardners, 2020-06-21:
That'd be a bug in the VHDL. use 138-hdmi-audio27mhz branch for now.  That should build.

frehwagen, 2020-06-21:
Actually what target are you trying to build? Looks like the Makefile lists unsupported targets with "all:". I'll fix this.

Currently we maintain mega65r2 and megaphoner1, if you want to try another then you need to specify this with you make command.

make bin/nexys4ddr.mcs

For instance.

rdpeake, 2020-06-21:
i'm just running make directly with no target - as i was just trying to get it to build to make sure i could use the subsytem rather than needing to use a vm, or seperate computer in advance of getting a dev kit
will use the branch gardners suggests

rdpeake, 2020-06-21:
unless frehwagen is correct in that i need to specify a different build target the 138-hdmi-audio-27mhz fails to build due to missing vhdl file /ipcore_dir/dotclock100.vhd.
i may need to clean, given the change in branch - but last time i used clean make failed to work again

Ben-401, 2020-06-21:
hi, thank you for the information on what dependancies are required for windows-subsystem-linnux.
I can use that info as I am updating the build instructions now, see #236 

How did you go in building the bitstream(s)?

What version of vivado are you using please?
i am just interested, i dont think there is a dependency on what version used...

rdpeake, 2020-06-21:
I have tried using `make` on branches stable-base, and 138-hdmi-audio-27mhz and they both failed during the build of 'mega65r1' for different reasons. based on frehwagen's comment i attempted to use mega65r2.mcs and megaphoner1.mcs on both branches however they all fail for the reason identified before. and as i indicated, development fails to build due to make errors. never tried master and figured it wasn't worth trying. the 'make clear' target may need to be double checked, as i tried to use that at one point and it didn't end well.

As i currently have no hardware to deploy this to, i'm not particularly worried about it not working, as i was mainly doing this to get it setup prior to the dev kit showing up some time later this year, and trying to understand how the code base is set up for potentially porting MISTer cores once i do get the dev kit.

rdpeake, 2020-06-21:
And because i forgot to answer - i am using vivado 2020.1

gardners, 2020-06-21:
Yes, I have only just updated to 2019.2.1 and don't currently have enough disk space to also install 2020.1 (its also a pain over satellite internet). It's possible that 2020.1 has some extra check that is detecting what it thinks is a multiple diver.  If it isn't a big hassle, you could try 2019.2.1 and see if the problems are still there?

rdpeake, 2020-06-21:
given i had to symlink 2020.1 in to the opt tool spot - it shouldn't be too hard to do it for 2019.2 will give it a go.

rdpeake, 2020-06-22:
using 2019.2 i was able to get to the make file trying to make mega65_ftp before it failed - given the statement above about what targets are maintained - i'm guessing this is a good place to be at and can wait until more details (and improvements are made) as dev kits get closer

Ben-401, 2020-06-22:
yes, we are working on cleaning up the build process.
the focus has been on development for a long while.
please come back soon and keep an eye out on us.

Ben-401, 2020-06-22:
its worth a mention, that the stable_base branch does in fact clone the Ophis
(see lines 30, 32, 42, 118, as shown below.
In that regard, I suggest this issue be closed, and the doco updated.
```
     1	devuser@u16vm:~/dev/mega65$ 
     2	devuser@u16vm:~/dev/mega65$ date
     3	Monday 22 June  11:03:20 ACST 2020
     4	devuser@u16vm:~/dev/mega65$ git clone https://github.com/mega65/mega65-core.git
     5	Cloning into 'mega65-core'...
     6	remote: Enumerating objects: 101, done.
     7	remote: Counting objects: 100% (101/101), done.
     8	remote: Compressing objects: 100% (77/77), done.
     9	remote: Total 45790 (delta 65), reused 49 (delta 24), pack-reused 45689
    10	Receiving objects: 100% (45790/45790), 25.95 MiB | 2.81 MiB/s, done.
    11	Resolving deltas: 100% (34509/34509), done.
    12	Checking connectivity... done.
    13	devuser@u16vm:~/dev/mega65$ cp -r mega65-core/ mega65-core-clean
    14	devuser@u16vm:~/dev/mega65$ cd mega65-core
    15	devuser@u16vm:~/dev/mega65/mega65-core$ git checkout stable_base 
    16	Branch stable_base set up to track remote branch stable_base from origin.
    17	Switched to a new branch 'stable_base'
    18	devuser@u16vm:~/dev/mega65/mega65-core$ git show
    19	commit 4a8842dbedf0eae360092dc603553da03c99bf2b
    20	Merge: a9ccc62 9ef3620
    21	Author: frehwagen <frehwagen@bluewaysw.de>
    22	Date:   Sun Jun 21 11:36:02 2020 +0200
    23	
    24	    Merge pull request #235 from Ben-401/stable_base
    25	    
    26	    update 2x submodules, change implementation of USE_LOCAL_CC65
    27	
    28	devuser@u16vm:~/dev/mega65/mega65-core$ make
    29	Makefile:106: =============================================================
    30	Makefile:106: ~~~~~~~~~~~~~~~~> Making: Ophis/bin/ophis
    31	git submodule init
    32	Submodule 'Ophis' (https://github.com/gardners/Ophis.git) registered for path 'Ophis'
    33	Submodule 'cbmconvert' (https://github.com/sasq64/cbmconvert.git) registered for path 'cbmconvert'
    34	Submodule 'cc65' (https://github.com/cc65/cc65.git) registered for path 'cc65'
    35	Submodule 'ghdl' (https://github.com/ghdl/ghdl.git) registered for path 'ghdl'
    36	Submodule 'iverilog' (https://github.com/steveicarus/iverilog.git) registered for path 'iverilog'
    37	Submodule 'src/mega65-fdisk' (https://github.com/MEGA65/mega65-fdisk.git) registered for path 'src/mega65-fdisk'
    38	Submodule 'src/mega65-freezemenu' (https://github.com/MEGA65/mega65-freezemenu.git) registered for path 'src/mega65-freezemenu'
    39	Submodule 'src/mega65-libc' (https://github.com/MEGA65/mega65-libc.git) registered for path 'src/mega65-libc'
    40	Submodule 'open-roms' (https://github.com/MEGA65/open-roms.git) registered for path 'src/open-roms'
    41	git submodule update
    42	Cloning into 'Ophis'...
    43	remote: Enumerating objects: 31, done.
    44	remote: Counting objects: 100% (31/31), done.
    45	remote: Compressing objects: 100% (19/19), done.
    46	remote: Total 1081 (delta 20), reused 23 (delta 12), pack-reused 1050
    47	Receiving objects: 100% (1081/1081), 3.36 MiB | 1.37 MiB/s, done.
    48	Resolving deltas: 100% (584/584), done.
    49	Checking connectivity... done.
    50	Submodule path 'Ophis': checked out 'e6ae193fb11ade39b9c7750b36f509da5106adc0'
    51	Cloning into 'cbmconvert'...
    52	remote: Enumerating objects: 42, done.
    53	remote: Total 42 (delta 0), reused 0 (delta 0), pack-reused 42
    54	Unpacking objects: 100% (42/42), done.
    55	Checking connectivity... done.
    56	Submodule path 'cbmconvert': checked out '1569e9c38d2873fc81c862d4f79b68cc51d2100f'
    57	Cloning into 'cc65'...
    58	remote: Enumerating objects: 110, done.
    59	remote: Counting objects: 100% (110/110), done.
    60	remote: Compressing objects: 100% (76/76), done.
    61	remote: Total 61702 (delta 54), reused 67 (delta 31), pack-reused 61592
    62	Receiving objects: 100% (61702/61702), 22.60 MiB | 2.49 MiB/s, done.
    63	Resolving deltas: 100% (39006/39006), done.
    64	Checking connectivity... done.
    65	Submodule path 'cc65': checked out '6c320f7d654d9c04482b3486abe063c1ac1735bd'
    66	Cloning into 'ghdl'...
[... 144 lines cut ...]
   211	Makefile:150: recipe for target '../wrk/common/xmalloc.o' failed
   212	make[2]: *** [../wrk/common/xmalloc.o] Interrupt
   213	Makefile:150: recipe for target '../wrk/common/filestat.o' failed
   214	make[2]: *** [../wrk/common/filestat.o] Interrupt
   215	Makefile:150: recipe for target '../wrk/common/filetype.o' failed
   216	make[2]: *** [../wrk/common/filetype.o] Interrupt
   217	make[2]: *** Deleting file '../wrk/common/cmdline.o'
   218	make[2]: *** wait: No child processes.  Stop.
   219	make[2]: *** Waiting for unfinished jobs....
   220	make[2]: *** wait: No child processes.  Stop.
   221	Makefile:150: recipe for target '../wrk/common/strutil.o' failed
   222	make[2]: *** [../wrk/common/strutil.o] Error 2
   223	make[2]: *** wait: No child processes.  Stop.
   224	make[2]: *** Waiting for unfinished jobs....
   225	make[2]: *** wait: No child processes.  Stop.
   226	Makefile:150: recipe for target '../wrk/common/hashtab.o' failed
   227	make[2]: *** [../wrk/common/hashtab.o] Error 2
   228	Makefile:6: recipe for target 'all' failed
   229	make[1]: *** [all] Interrupt
   230	Makefile:95: recipe for target 'cc65/bin/cc65' failed
   231	make: *** [cc65/bin/cc65] Interrupt
   232	
   233	devuser@u16vm:~/dev/mega65/mega65-core$ 
   234	
devuser@u16vm:~/dev/ben-401/mega65-core$

rdpeake, 2020-06-22:
yes, the original issue was resolved once the correct branch to build was given - also able to build supported targets using 2019.2.1 of vivado and the listed dependencies on WSL.

closed 2020-06-22 by rdpeake