MEGA65/mega65-core #242: old vivado 2016.2 unsupported, resolve dependency (u16), xpr tcl dsa.num_compute_units

Ben-401, 2020-06-22:
in ubuntu 16 with vivado 2016.2:
```make``` fails, as does ``make bin/nexys4ddr.bit```.

Seems the TCL-script that creates the XPR-file uses a setting that is not supported in v2016.2.
I am yet to confirm this.

```
devuser@u16vm:~/dev/ben-401/mega65-core$ make
Makefile:823: =============================================================
Makefile:823: ~~~~~~~~~~~~~~~~> Making: vivado/mega65r1.xpr
echo MOOSE vivado/mega65r1.xpr from vivado/mega65r1_gen.tcl
MOOSE vivado/mega65r1.xpr from vivado/mega65r1_gen.tcl
./vivado_wrapper -mode batch -source vivado/mega65r1_gen.tcl
vivado -mode batch -source vivado/mega65r1_gen.tcl

****** Vivado v2016.2 (64-bit)
  **** SW Build 1577090 on Thu Jun  2 16:32:35 MDT 2016
  **** IP Build 1577682 on Fri Jun  3 12:00:54 MDT 2016
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source vivado/mega65r1_gen.tcl
# set origin_dir [file dirname [info script]]/..
# if { [info exists ::origin_dir_loc] } {
#   set origin_dir $::origin_dir_loc
# }
# set project_name "mega65r1"
# if { [info exists ::user_project_name] } {
#   set project_name $::user_project_name
# }
# variable script_file
# set script_file "mega65r1.tcl"
# proc help {} {
#   variable script_file
#   puts "\nDescription:"
#   puts "Recreate a Vivado project from this script. The created project will be"
#   puts "functionally equivalent to the original project for which this script was"
#   puts "generated. The script contains commands for creating a project, filesets,"
#   puts "runs, adding/importing sources and setting properties on various objects.\n"
#   puts "Syntax:"
#   puts "$script_file"
#   puts "$script_file -tclargs \[--origin_dir <path>\]"
#   puts "$script_file -tclargs \[--project_name <name>\]"
#   puts "$script_file -tclargs \[--help\]\n"
#   puts "Usage:"
#   puts "Name                   Description"
#   puts "-------------------------------------------------------------------------"
#   puts "\[--origin_dir <path>\]  Determine source file paths wrt this path. Default"
#   puts "                       origin_dir path value is \".\", otherwise, the value"
#   puts "                       that was set with the \"-paths_relative_to\" switch"
#   puts "                       when this script was generated.\n"
#   puts "\[--project_name <name>\] Create project with the specified name. Default"
#   puts "                       name is the name of the project from where this"
#   puts "                       script was generated.\n"
#   puts "\[--help\]               Print help information for this script"
#   puts "-------------------------------------------------------------------------\n"
#   exit 0
# }
# if { $::argc > 0 } {
#   for {set i 0} {$i < [llength $::argc]} {incr i} {
#     set option [string trim [lindex $::argv $i]]
#     switch -regexp -- $option {
#       "--origin_dir"   { incr i; set origin_dir [lindex $::argv $i] }
#       "--project_name" { incr i; set project_name [lindex $::argv $i] }
#       "--help"         { help }
#       default {
#         if { [regexp {^-} $option] } {
#           puts "ERROR: Unknown option '$option' specified, please type '$script_file -tclargs --help' for usage info.\n"
#           return 1
#         }
#       }
[... 104 lines cut ...]
#     set option [string trim [lindex $::argv $i]]
#     switch -regexp -- $option {
#       "--origin_dir"   { incr i; set origin_dir [lindex $::argv $i] }
#       "--project_name" { incr i; set project_name [lindex $::argv $i] }
#       "--help"         { help }
#       default {
#         if { [regexp {^-} $option] } {
#           puts "ERROR: Unknown option '$option' specified, please type '$script_file -tclargs --help' for usage info.\n"
#           return 1
#         }
#       }
#     }
#   }
# }
# create_project -force ${project_name} vivado/ -part xc7a100tcsg324-1
# set proj_dir [get_property directory [current_project]]
# set obj [current_project]
# set_property -name "default_lib" -value "xil_defaultlib" -objects $obj
# set_property -name "dsa.num_compute_units" -value "60" -objects $obj
ERROR: [Common 17-142] Invalid property name 'dsa.num_compute_units'.  Please use the 'list_property' command to find properties supported by the target 'project' object.
INFO: [Common 17-206] Exiting Vivado at Mon Jun 22 17:08:08 2020...
Makefile:823: recipe for target 'vivado/nexys4ddr.xpr' failed
make: *** [vivado/nexys4ddr.xpr] Error 1
devuser@u16vm:~/dev/ben-401/mega65-core$
```

Ben-401, 2020-06-22:
i will try vivado 2018.2.1 in ubuntu 16.

Should all targets in the "./vivado/*" directory work?
I will assume that the targets in that dir (on stable_base) of interest are:
mega65r2
nexys4ddr
nexys4

Can someone please advise?

gardners, 2020-06-22:
For now, assume only mega65r2 and megaphoner1 work, and maybe nexys4ddr-widget

Ben-401, 2020-06-22:
Thanks for that @paul,
I have confirmed that with vivado 2018.2,
```make``` generates ```bit/mega65r1.bit``` but fails when contuining on to ```bit/nexys4ddr.bit```.
error log is below.

i have started specifying targets now, and will verify those two/three you listed, 

below is the log showing that the ```make``` (ie make all), after the mega65r1 was built, continued onto make the nexys4ddr, but failed.
I will not look into why it fails, merely show that it does fail.
```
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a100t'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a100t'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 7128 
---------------------------------------------------------------------------------
Starting RTL Elaboration : Time (s): cpu = 00:00:02 ; elapsed = 00:00:03 . Memory (MB): peak = 1319.797 ; gain = 97.914 ; free physical = 4832 ; free virtual = 7511
---------------------------------------------------------------------------------
ERROR: [Synth 8-2032] formal clock80 is not declared [/home/devuser/dev/ben-401/mega65-core/src/vhdl/nexys4ddr.vhdl:319]
ERROR: [Synth 8-2778] type error near qspidb ; expected type unsigned [/home/devuser/dev/ben-401/mega65-core/src/vhdl/nexys4ddr.vhdl:346]
ERROR: [Synth 8-2032] formal clock240 is not declared [/home/devuser/dev/ben-401/mega65-core/src/vhdl/nexys4ddr.vhdl:396]
ERROR: [Synth 8-2029] formal clock50mhz has no actual or default value [/home/devuser/dev/ben-401/mega65-core/src/vhdl/nexys4ddr.vhdl:388]
INFO: [Synth 8-994] clock50mhz is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:60]
INFO: [Synth 8-994] clock100 is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:61]
INFO: [Synth 8-994] clock27 is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:62]
INFO: [Synth 8-994] clock162 is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:63]
INFO: [Synth 8-994] btncpureset is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:66]
INFO: [Synth 8-994] irq is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:68]
INFO: [Synth 8-994] nmi is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:69]
INFO: [Synth 8-994] restore_key is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:70]
<snip>
INFO: [Synth 8-994] btn is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:327]
INFO: [Synth 8-994] rsrx is declared here [/home/devuser/dev/ben-401/mega65-core/src/vhdl/machine.vhdl:330]
INFO: [Synth 8-2810] unit behavioral ignored due to previous errors [/home/devuser/dev/ben-401/mega65-core/src/vhdl/nexys4ddr.vhdl:173]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:03 ; elapsed = 00:00:05 . Memory (MB): peak = 1486.422 ; gain = 264.539 ; free physical = 4853 ; free virtual = 7533
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
65 Infos, 0 Warnings, 0 Critical Warnings and 5 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Mon Jun 22 19:40:11 2020...
[Mon Jun 22 19:40:11 2020] synth_1 finished
wait_on_run: Time (s): cpu = 00:00:13 ; elapsed = 00:00:14 . Memory (MB): peak = 1230.836 ; gain = 0.000 ; free physical = 5162 ; free virtual = 7841
# launch_runs impl_1 -to_step write_bitstream
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Mon Jun 22 19:40:11 2020...
Makefile:831: recipe for target 'bin/nexys4ddr.bit' failed
make: *** [bin/nexys4ddr.bit] Error 1
devuser@u16vm:~/dev/ben-401/mega65-core$
```

gardners, 2020-06-22:
It's fine if we specify a minimum version of Vivado, e.g., of 2019.2.1, as there are known bugs in earlier versions.

closed 2022-01-03 by lydon42