MEGA65/mega65-core #244: add license/copyright info as per FSFE recommendations using REUSE

Ben-401, 2020-06-28:
Currently there seems only one license artifact, namely ```./LICENSE.txt```.

It is suggested that this project adopt the FSFE recommendations and implement the REUSE specification:
https://fsfe.org/activities/ftf/faq-what-is-licensing.en.html
https://reuse.software/tutorial/

I also note the existence of the ```TRADEMARKS.md``` file.

Ben-401, 2020-06-29:
I have created a script-v1 to implement some of this.

On branch 138-hdmi-audio-27mhz
before any mods are done, the ```reuse lint``` reports:

# SUMMARY

* Bad licenses:
* Deprecated licenses:
* Licenses without file extension:
* Missing licenses:
* Unused licenses:
* Used licenses: CC0-1.0, LGPL-3.0-or-later
* Read errors: 0
* Files with copyright information: 64 / 532
* Files with license information: 0 / 532

Unfortunately, your project is not compliant with version 3.0 of the REUSE Specification :-(

Ben-401, 2020-06-29:
now, after applying the script above, reuse-lint reports:

# SUMMARY

* Bad licenses:
* Deprecated licenses:
* Licenses without file extension:
* Missing licenses:
* Unused licenses:
* Used licenses: CC0-1.0, LGPL-3.0-or-later
* Read errors: 0
* Files with copyright information: 74 / 532
* Files with license information: 10 / 532

Unfortunately, your project is not compliant with version 3.0 of the REUSE Specification :-(

Ben-401, 2020-07-13:
please note the WARNING in commit 7bb04f7, regarding copyright of some fonts and PRGs (ie boulderdash).
this warning is in the commit second to last above.

gardners, 2020-07-13:
The graphical elements in assets are all originals, and can stay.
Bouldermark and synthmark are publicly distributable binaries. But feel free to remove those two from the repo to avoid all potential problems.

Ben-401, 2020-07-14:
now when running the script (implementing BIN and SCR files),
```reuse lint``` reports:
* Bad licenses:
* Deprecated licenses:
* Licenses without file extension:
* Missing licenses:
* Unused licenses:
* Used licenses: CC0-1.0, LGPL-3.0-or-later
* Read errors: 0
* Files with copyright information: 123 / 541
* Files with license information: 61 / 541

The bulk of the files will be processed in the TXT section, to be worked on next.

Ben-401, 2020-07-20:
is now ready for review and update of LICENSEHEADER and LICENSEFOOTER
Also needs update of the normalise.data-file

gardners, 2020-08-11:
We should proceed with applying this to the 138-hdmi-audio-27mhz branch.
What is the process for running your scripts, so that I can verify that all is well?

Ben-401, 2020-08-11:
you need a clean checkout of the repo, change to desired branch
(this is because you dont want to tag build artefacts)

NOTE that this script was developed/customised for the master branch.

copy in the following files to the root dir (from Ben-401/mega65-core/test-license2):
license-parse.sh
license-parse-norm.dat

run the "./licence-parse" and observe the text output.

at the top of the script there is a LICMODE variable that controls what of the three 'types' of files to process. This is so that just one of the three 'types' is processed at a time. It is suggested to run/verify each type first, then run/verify/add/commit just one 'type' for each 'type' so that the commits are seperated.

by default, as the script runs and processes one 'type', the changed/created file is added to git for easy workflow, but then followed by resetting and/or removing/undoing those changes, effectively performing a DRYRUN.

Toward the bottom of each 'type' section are comments "DRYRUN". When you are confident that the script correctly processes the files, comment out the lines under DRYRUN. This will stage the changes ready for committing.

Ben-401, 2020-08-11:
this script generates numerous files, all prepended with LICPARSE (or similar).

Ben-401, 2020-08-11:
Regarding the contents of "license-parse-norm.dat",
I was going to suggest an email be sent to all variants of the same-user, requesting advise on which they prefer,
we could choose what we think is best, and advise them if failing to hear back, we will proceed with our choice.

lydon42, 2024-08-17:
Can someone bring me up to date on this?
Is this "adding proper license headers to all source files using a script"?