MEGA65/mega65-core #247: MEGA65 does not boot if an external POWERED OFF Disc Drive is connected to IEC

adtbm, 2020-07-02:
If an external drive 1581/1541-II is connected to the MEGA65 but is not powered on, the MEGA65 will not boot into C65, it does boot into C64, when MEGA-Key is pressed while switching MEGA65 on.

gardners, 2020-07-02:
This will be due to the powered-off drive holding one or more of the IEC lines low, which the C65 DOS misinterprets as there being a device connected.  Please try making an IEC cable with pull-up resistors to see if this resolves the problem, while not stopping normal IEC operation. If so, we should add/strengthen the IEC pull-ups on the M65 mainboard.

gardners, 2020-07-02:
Confirmed M65 R2 lacks pullups, but C64 has them.

RP6 places 1K pullup to 5V on clock, data and ATN:

http://www.zimmers.net/anonftp/pub/cbm/schematics/computers/c64/250469-rev.B-left-corrected2.gif

C128 also puts pull-ups on the SRQ line via RP5.4:

http://www.zimmers.net/anonftp/pub/cbm/schematics/computers/c128/310378-3-left.gif

We need to make sure that Antti is adding these in the list of missing pullups on the R3 PCB

deftmega, 2021-02-28:
The problem is not completely gone with R3. If 1541-II is connected over IEC and turned off, machine will do (tested with DevKit release default disk image): 
a) Lock-up during autoboot in C65 Basic
b) Ignore full speed (40 Mhz) switch
c) Randomly lock up in C65 Mode

dansanderson, 2025-03-31:
Related: https://github.com/MEGA65/mega65-rom-public/issues/169

The pending HW IEC implementation does not have this issue.