MEGA65/mega65-core #252: Powering on MEGA65 without SD card gives BREAK AT message

adtbm, 2020-07-13:
With the latest 138-hd-5503 bitstream, when switching on MEGA65 without an SD card inserted, MEGA65 will come up with Blue screen and a BREAK AT message...
[image]

gardners, 2020-07-13:
This is caused by an interrupt happening before the interrupt vectors get setup.
I'll take a look into this.

FeralChild64, 2020-07-14:
The start does not go trough the $FFFC vector, right? Otherwise you would have SEI as the very first instruction executed. I guess there must be some garbage at $314 - if there was a $0000, the Kernal routine would have executed internal IRQ handling directly. Another possibility is the KEYLOG vector ($28F) - it is used by the IRQ keyboard scanning routine, it has to be initialized properly.

lydon42, 2022-01-04:
Can't reproduce.

closed 2022-10-02 by lydon42