MEGA65/mega65-core #253: SD cards MBR is getting overwritten so often

adtbm, 2020-07-13:
it happens quite often, that the MBR is getting killed on the SD card.

Pauls quote: 
"writing over sector 0 is also the most likely brain fade to happen with random access to the SD card.
I should make it so that there is a special sequence required to write the MBR since it is so important."

Ben-401, 2020-07-13:
by killed do you mean overwritten? with zeros or garbage? just one byte or many?
and on what branch exactly, as I dont have a "138-hd-5503" branch as reported in issue #252.

gardners, 2020-07-13:
What Anton is talking about implementing here is a safety catch, so that you can't easily cause the MBR to be overwritten.  The problem is branch independent: Basically if you write $03 to $D680, when no sector number has been supplied, then it will write over the MBR, e.g., if some program writes over the IO space without really paying attention.  This is also why in the longer term direct SD card access will be disabled.

Ben-401, 2020-08-20:
bump - just testing the "https://github.com/MEGA65/mega65-test-procedures"

lydon42, 2022-10-02:
Is this still a problem?

lydon42, 2024-08-17:
Possible related to (and perhaps also fixed by) #264 ?

gardners, 2024-08-17:
Yes, fixed by #264. Evidence for this is that noone has complained for several years about this happening,

closed 2024-08-17 by gardners