MEGA65/mega65-core #254: Hardware Sprite Collision Detection - Feature Request

stepzC64, 2020-07-14:
hey, as discussed on Discord with you, if possible could you consider adding a way so that the original C64 hardware sprite collision is extended with a new feature which allows you determine, when more than 2 hardware sprites are colliding, which individual sprites are colliding with which other individual sprites, thank you

lydon42, 2022-01-04:
What would be the way to implement that? One collision register per sprite? That is lot of space...

Or you can set one sprite for which you will get a extra collision register? so essentially you can make one sprite your player sprite and that will register collisions in an extra register. Could be also combined with background collision... and if fixed to sprite 0 would not require an extra way to specify the "special "sprite.
Then bit 0 could be background collision with sprite 0 and bit 1-7 are collisions with the other 7 sprites.

If there is enough place for all 8 sprite/sprite collisions, the bit representing the sprite itself could be made the background collision. Again... do we have space for 8 extra collision registers?

gardners, 2024-08-17:
I like the idea, but it would be tricky indeed, due to the lack of spare VIC-IV registers.  Thus I don't think that it is feasible to implement.

That said, enabling sprite collision interrupts and the speed of the MEGA65 should allow for responding to each collision as it occurs.

closed 2024-08-17 by lydon42