MEGA65/mega65-core #255: monitor_load and nexys4ddr core build errors

vvrbanc, 2020-07-14:
On branch master/stable_base commit 4a8842dbedf0eae360092dc603553da03c99bf2b.

I ran into some issues:

**1) src/tools/monitor_load.c compile fails with multiple errors:**

src/tools/monitor_load.c:152:1: error: unknown type name ‘uint32_t’; did you mean ‘u_int32_t’?

FIX: add #include <stdint.h>

undeclared variables/pointers:

screen_address
next_screen_address
screen_width
screen_line_step
screen_rows_remaining
screen_line_offset
screen_line_buffer

undefined function 'print_screencode'

FIX: I'm guessing some files are missing from source control? I don't see these declared anywhere in the entire repository.

**2) nexys4ddr.mcs target fails during synthesis:**

ERROR: [Synth 8-2032] formal clock80 is not declared [/var/tmp/mega65-core/src/vhdl/nexys4ddr.vhdl:319]
ERROR: [Synth 8-2778] type error near qspidb ; expected type unsigned [/var/tmp/mega65-core/src/vhdl/nexys4ddr.vhdl:346]
ERROR: [Synth 8-2032] formal clock240 is not declared [/var/tmp/mega65-core/src/vhdl/nexys4ddr.vhdl:396]
ERROR: [Synth 8-2029] formal clock50mhz has no actual or default value [/var/tmp/mega65-core/src/vhdl/nexys4ddr.vhdl:388]

It seems to me like some refactoring was done and the nexys4ddr.vhdl was not updated to reflect these changes.

A lot of clock lines were changed/renamed.

Nexys4 (non-ddr) builds successfully though. About a year back these two files (nexys4.vhdl and nexys4ddr.vhdl) were identical so I just copied nexys4.vhdl over nexys4ddr.vhdl and changed the line 390 to "target => nexys4ddr".

Looks like LCD panel output (for pmod?) is not present in nexys4.vhdl so that was lost in the process. It used some of the changed clock lines and I didn't know how to fix this. 

Either way, I can't test lcd panel output since I don't have that pmod. Ideally nexys4ddr.vhdl should be updated, and the LCD panel part merged back into nexys4.vhdl? 

**3) more build errors for nexys4ddr.mcs:**

vivado/nexys4ddr_gen.tcl seems out of date, and nexys4_gen.tcl seemed more up to date so I merged over what made sense.

I've uploaded my fixes here: https://github.com/vvrbanc/mega65-core/tree/vv005 . I advise reviewing the changes before merging, I have no idea what I'm doing :) FWIW it compiles and runs.
commit is here: https://github.com/vvrbanc/mega65-core/commit/60bf44d5050aba12d361eace8a5049d1615ed23b

gardners, 2020-07-14:
For (1), monitor_load.c should just be taken out of this repo, as it has been subsumed by m65.c in the github.com/mega65/mega65-tools repo.

For (2) and (3), we currently only support the nexys4ddr-widget target for the Nexys4 boards.  We should probably just retire the nexys4ddr target.  But I have no objection to it being resurrected, either, if you have it working again.

vvrbanc, 2020-07-14:
Thanks, I didn't realize.

I think it is better to retire nexys4ddr non-widget target.

I suppose all that needs to be done then is replacing nexys4ddr with nexys4ddr-widget in Makefile, as well as removing monitor_load files and Makefile target.

closed 2020-07-14 by vvrbanc