MEGA65/mega65-core #256: Not possible to enter Configure, when using microSD card

adtbm, 2020-07-14:
when trying to access the MEGA65 configure menu during startup, with a microSD card inserted, MEGA65 only brings black screen.
Using the same card in the internal SD card slot, entering Configure works.

gardners, 2020-07-16:
The problem here is that the configuration menu is launched BEFORE Hyppo probes the SD cards, so we don't know
on which SD bus to look for the config sector.

We could make the configure programme probe the busses first.  

Longer term, it makes sense to preferentially read/write config sector to NVRAM, on those models that have it.  But SD card should still be a fall-back.

lydon42, 2022-10-02:
int/ext SD card switching does work with recent cores. It could still be enhanced, I think, but the bug mentioned here can't be reproduced anymore.

closed 2022-10-02 by lydon42