MEGA65/mega65-core #259: suggest delete ./ipcore_dir as no longer required

Ben-401, 2020-07-15:
this dir contains at least one source file, listed below.
otherwise, I dont think we need all these files anymore.

=> Are you OK with a PR to delete?

in the master/stable_base, i can see that the ```dotclock100.vhd``` is used,
so this file could be moved to ```src/vhdl``` before.

Ben-401, 2020-07-15:
similarly the ```./isework``` dir is legacy and can also be removed.

Ben-401, 2020-07-15:
similarly the ```./_unused/``` and ```./_unused2/``` could also go.
They have been here for years...

gardners, 2020-07-15:
Hello,

The dotclock100 was generated by ipcore stuff, so has the licence
encumberment of that.  Note that in 138-hdmi-audio-27mhz, it isn't needed
any more, so it is on its way out, anyway.

Paul.

closed 2022-10-02 by lydon42