MEGA65/mega65-core #260: Convert HYPPO to use open-source ACME assembler

gardners, 2020-07-17:
KickAss is great, but its closed source. Also, it is written in JAva, so can't (eventually) be ported to run native on the MEGA65.  Finally, ACME has 1st class support for all MEGA65 CPU features, so it makes it a logical choice.

gardners, 2020-07-17:
Now boots to BASIC 10 and does disk mounting etc with ACME-compiled HICKUP.
Ready for staging for merging.

Ben-401, 2020-08-08:
In branch 138-hdmi-audio-27mhz, the use of the ACME compiler is used in place of the KickAss.
But how to install ACME? can it be a submodule from github?
And should that branch also remove the KickAss files? i think so

Ben-401, 2020-08-08:
https://github.com/meonwax/acme
https://sourceforge.net/projects/acme-crossass/

gardners, 2020-08-08:
Yes, by all means produce a pull request adding auto-compilation of ACME

lydon42, 2021-12-21:
KickAss is removed and no longer used, HYPPO uses acme.

closed 2021-12-21 by lydon42