MEGA65/mega65-core #267: VIC-IV: SPRPTRBNK reset when legacy registers are touched

hernandp, 2020-08-06:
When registers considered as "legacy" are touched and HOTREG is enabled,  sprite pointer address registers (SPRPTRADR) are modified accordingly, but  SPRPTRBANK (0..6 bits of $D06E) does not seem to be set. This is correct? Do we need to reset it to zero? 

https://github.com/MEGA65/mega65-core/blob/b728aa8ea375f39449b8cccf3e12e9413227ada3/src/vhdl/viciv.vhdl#L1482

gardners, 2020-08-06:
To be tested when synthesis runs, then can move to staging.

gardners, 2020-09-11:
Related to this, the banks are being set incorrectly, as VIC-IV looks at writes to $DD00, but not the DDR register at $DD02 when picking the VIC-II bank.  It needs to get the cooked pin voltages instead.

lydon42, 2022-01-05:
This seems to be somewhat inconsistent with the handling auf CHARPTR in #368:
Here the MSB is *not* reset to allow for use of high memory area in legacy software.

closed 2022-10-02 by lydon42