MEGA65/mega65-core #269: inconsistencies across branches, -> please merge 138-* into master

Ben-401, 2020-08-07:
There are numerous differences between the two main/active branches, namely:
- master/stable_base, and
- 138-hdmi-audio-27mhz
These differences include change of filenames, obvious code, tool chain, etc.

Can we please resolve these consistencies by merging the two branches above??

Ben-401, 2020-08-07:
current problem with 138-hdmi-audio-27mhz is that it does not build 'out-of-the-box'.
These building issues have been resolved in master.

current problem with master/stable_base is that it does not support the new features that is being actively developed in 138-*

Can we please all get onto the same page? so that we can have the best of both worlds?

gardners, 2020-08-07:
the 138 branch implemented a number of important changes to clocking and hdmi output etc, that are required.

What is required to merge the branches is to identify what regressions the 138 branch has compared to stable_base, and fix those, so that we can advance stable_base to the head of the 138 branch.

That is, getting the test framework up and running is the key to advancing stable_base.

lydon42, 2022-01-05:
Current pre relase development has everything from 138-hdmi-audio-27mhz, as a compare shows.

closed 2022-01-05 by lydon42